diyhue/Lights #2: Generic_Dimmable_Light pin can not be changed

PowderK, 2018-12-26:
hy I use the Generic_Dimmable_Light with the following board.

https://www.esp8266.com/viewtopic.php?f=13&t=2929

  For this I need a control of the GPIO 13. my setup looks like this.

> #define
> 
>  use_hardware_switch false // To control on / off state and brightness using GPIO / pushbutton, set this value to true.
> // For GPIO based on / off and brightness control, the following is GPIO pins to ground using 10k resistor
> #define button1_pin 4 // on and brightness up
> #define button2_pin 5 // off and brightness down
> 
> // define pins
> #define LIGHTS_COUNT 1
> uint8_t pins [LIGHTS_COUNT] = {13};

However, the GIPO 0 is addressed. no matter which pin I specify, only the GPIO 0 will be activated.

Mevel, 2018-12-27:
I think the reference to the pin array is missing in  `analogWrite(i, (int)(current_bri[i]));`

I think this could work `analogWrite(pins[i], (int)(current_bri[i]));`

mariusmotea, 2018-12-27:
Exactly @Mevel .

@PowderK please test now with the new commit i made.

PowderK, 2018-12-29:
It works perfectly thanks for the help.

closed 2018-12-29 by PowderK