diyhue/Lights #6: ESP8266 Upload Error

charlescbullock, 2019-01-21:
First off, I wanted to say THANK YOU to you guys helping me out!
**I would assume my issue is coming from not properly flashing the firmware?
I'm using WS2812B LED strips, and a Raspberry Pi 3 B+ as a DIY Hue emulator.**

Here is a link to the ESP8266 I have: https://www.amazon.com/gp/product/B010N1SPRK/ref=ppx_yo_dt_b_asin_title_o08__o00_s00?ie=UTF8&psc=1

These are the steps I followed to upload your firmware:

**1. Download entire folder at: https://github.com/diyhue/Lights
2. Plug in ESP8266 Device
3. Open Arduino IDE
4. Within IDE, open Board Manager and search for ESP8266.
5. Download only option, and select it as board.
6. Confirm COM Port of board, and set that within Arduino IDE.
7. Set ESP8266 Upload Speed to 115200 within Arduino IDE.
8. Plug in USB-UART Adapter for troubleshooting, and run it through Putty at a speed of 115200.
9. In Arduino IDE, open Sketch file "WS2812BHueStrip.ino" located here: https://github.com/diyhue/Lights/tree/master/Arduino/WS2812BHueStrip
10. Uncomment and change IP settings to allow Static IP Address of 192.168.1.201, and the Router set to 192.168.1.1.
11. set "lightsCount" to 5 and set "pixelCount" to 30
12. Resave Sketch as date/time based name in Arduino Documents folder
13. Verify
14. Upload**

**So, in Putty - I'm constantly seeing the ESP8266 search through the WiFi SSID's in the area (shown below):**

`scan start
scan done
21 networks found
1: word56-42 (-72)*
2: xfinitywifi (-74)
3: NETGEAR36 (-81)*
4: NVR9ca3a94eb62f (-89)*
5: NETGEAR24 (-85)*
6: HailtotheVictors (-90)*
7: Harley (-86)*
8: PS4-59EB415741A1 (-40)*
9: CraigFam (-85)*
10: Harley (-89)*
11: Harley (-74)*
12: tl60fdbe (-86)*
13: ATT4bxJ7ZC (-88)*
14: Aperture Laboratories (-80)*
15: Aperture Innovators (-78)*
16: ATTBQMjuwa (-86)*
17: xfinitywifi (-89)
18: Blips and Chitz (-40)*
19: dbfcfbcbead5e3e456e1257d6bd01474
                                    ▒ (-60)*
20: Harley (-76)*
21: HP-Print-8C-Deskjet 2540 series (-91)*

scan start
scan done
20 networks found
1: word56-42 (-72)*
2: xfinitywifi (-74)
3: xfinitywifi (-91)
4: NETGEAR36 (-83)*
5: NETGEAR24 (-86)*
6: 68741b5a6977bc8ad260e494d5c59971▒ (-91)*
7: HailtotheVictors (-93)*
8: PS4-59EB415741A1 (-52)*
9: CraigFam (-85)*
10: Harley (-84)*
11: Harley (-73)*
12: tl60fdbe (-82)*
13: Aperture Laboratories (-78)*
14: Aperture Innovators (-77)*
15: NVR9ca3a94eb62f (-90)*
16: Blips and Chitz (-53)*
17: dbfcfbcbead5e3e456e1257d6bd01474
                                    ▒ (-60)*
18: ATTmuYDzMi (-93)*
19: Harley (-77)*
20: ATTgZWCFPi (-91)*
`

**Here is the Error Message copied from Arduino IDE:** 

`Arduino: 1.8.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Enabled, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Sketch + WiFi Settings, 256000"

Build options changed, rebuilding all
Sketch uses 434640 bytes (86%) of program storage space. Maximum is 499696 bytes.
Global variables use 36464 bytes (44%) of dynamic memory, leaving 45456 bytes for local variables. Maximum is 81920 bytes.
C:\Users\charl\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9/esptool.exe -vv -cd ck -cb 256000 -cp COM3 -ca 0x7C000 -cz 0x4000 -ca 0x00000 -cf C:\Users\charl\AppData\Local\Temp\arduino_build_957736/sketch_jan21a.ino.bin 
esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 256000
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x0007C000
	setting erase size to 0x00004000
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
An error occurred while uploading the sketch
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 1E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 1E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 9E instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 1E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
warning: espcomm_sync failed
error: espcomm_open failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

**Thank you all, again, for any help!!**

charlescbullock, 2019-01-21:
Thanks for a speedy response!!!
And thank you for the knowledge. That is good to know!

But, I have had no luck either way.

I just tried the method you had suggested, and I still got this set of error messages:

`Arduino: 1.8.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Enabled, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Sketch + WiFi Settings, 256000"

Sketch uses 434848 bytes (87%) of program storage space. Maximum is 499696 bytes.
Global variables use 36496 bytes (44%) of dynamic memory, leaving 45424 bytes for local variables. Maximum is 81920 bytes.
C:\Users\charl\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9/esptool.exe -vv -cd ck -cb 256000 -cp COM3 -ca 0x7C000 -cz 0x4000 -ca 0x00000 -cf C:\Users\charl\AppData\Local\Temp\arduino_build_435350/sketch_jan21a.ino.bin 
esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 256000
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x0007C000
An error occurred while uploading the sketch
	setting erase size to 0x00004000
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 1E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 78 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 1E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
[... 832 lines cut ...]
              strip.SetPixelColor(pixel + transitionLeds / 2 + lightLedsCount * light, blendingEntert(rgb[light], rgb[light + 1], pixel - (lightLedsCount - transitionLeds ) + 1));
            }
          }
        }
      } else {
        strip.ClearTo(RgbColor(rgb[0][0], rgb[0][1], rgb[0][2]), 0, lightLedsCount - 1);
      }
    }
    strip.Show();
  }
}

void loop() {
  ArduinoOTA.handle();
  server.handleClient();
  if (!entertainment_run) {
    lightEngine();
  } else {
    if ((millis() - lastEPMillis) >= entertainmentTimeout) {
      entertainment_run = false;
      restore();
    }
  }
  entertainment();
}`

charlescbullock, 2019-01-21:
**Thanks for a speedy response!!!
And thank you for the knowledge. That is good to know!

But, I have had no luck either way.

I just tried the method you had suggested, and I still got this set of error messages:**

`Arduino: 1.8.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Enabled, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Sketch + WiFi Settings, 256000"

Sketch uses 434848 bytes (87%) of program storage space. Maximum is 499696 bytes.
Global variables use 36496 bytes (44%) of dynamic memory, leaving 45424 bytes for local variables. Maximum is 81920 bytes.
C:\Users\charl\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9/esptool.exe -vv -cd ck -cb 256000 -cp COM3 -ca 0x7C000 -cz 0x4000 -ca 0x00000 -cf C:\Users\charl\AppData\Local\Temp\arduino_build_435350/sketch_jan21a.ino.bin
esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel ck@atelier-klippel.de
setting board to ck
setting baudrate from 115200 to 256000
setting port from COM1 to COM3
setting address from 0x00000000 to 0x0007C000
An error occurred while uploading the sketch
setting erase size to 0x00004000
setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
read 0, requested 1
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
serialport_receive_C0: 1E instead of C0
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
serialport_receive_C0: 78 instead of C0
resetting board
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
read 0, requested 1
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
serialport_receive_C0: 1E instead of C0
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
[... 831 lines cut ...]
strip.SetPixelColor(pixel + transitionLeds / 2 + lightLedsCount * light, blendingEntert(rgb[light], rgb[light + 1], pixel - (lightLedsCount - transitionLeds ) + 1));
}
}
}
} else {
strip.ClearTo(RgbColor(rgb[0][0], rgb[0][1], rgb[0][2]), 0, lightLedsCount - 1);
}
}
strip.Show();
}
}

void loop() {
ArduinoOTA.handle();
server.handleClient();
if (!entertainment_run) {
lightEngine();
} else {
if ((millis() - lastEPMillis) >= entertainmentTimeout) {
entertainment_run = false;
restore();
}
}
entertainment();
}`

cheesemarathon, 2019-01-21:
There are two buttons on your board, a flash button, and a reset button. If the board starts and the flash button is held, it puts the board into flashing mode. This allows code to be uploaded. Some boards will enter this mode automatically when the upload starts, some won't. To enter this mode yourself press and hold the flash button and then press and release the RST button to restart the board. The board is now in flash mode. I have found some boards dont stay in this mode for long, so I recomend putting it into flash mode quickly, right after you see these lines at the bottom of the arduino IDE:

```
Arduino: 1.8.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Enabled, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Sketch + WiFi Settings, 256000"
```

charlescbullock, 2019-01-21:
Awesome!
That made a successful upload!!!
However, I'm not able to find the Light in the Hue app.. Any ideas?

Thank you again!!!!!!!

charlescbullock, 2019-01-22:
> When I search, it says in the app it was unable to find any lights but 30
> seconds later they appear, so make sure you wait long enough.
>
> If not follow the instructions here
> <https://diyhue.readthedocs.io/en/latest/AddFuncts/debug.html#host-install>
> to enable debug mode and post the output when you try to search.

Am I supposed to configure something in the WiFiManager.cpp file, to put
the ESP8266 on my network?

That might be the issue I'm running into.

I tried the method you had suggested, waiting for the device to appear, and
didn't have luck.

I tried to use the USB-UART device to look into what was going on, and it
wrote out a mesh of about 10-20 characters in one line... Looked like
encrypted data.. But really not much.

The WS2812B lights were outputting an intensity chase in a red color, from
the first light to somewhere in the middle of the strip.

mariusmotea, 2019-01-22:
I recommend to try following things:

- wipe entire flash with `esptool.py erase_flash`
- install esp8266 library 2.4.2
- select board nodemcu v1.0
- libraries: wifi manager 0.14.0, neopixelbus by makuna 2.4.1

disconnect serial adapter and look with wifi network with the name set in sketch header (defaul is hue strip ....)

charlescbullock, 2019-01-23:
Awesome!  Thank you all, so much!!

I was able to get a strip up and running!

However, I was not able to get the ESP8266 onto my network successfully
with a static IP address.

Is that a functional property of the current sketch for the WS2812B Hue
Strip?

Once I recommented the line " //#define USE_STATIC_IP //! uncomment to
enable Static IP Adress"  it worked fine.

It would be great if I were able to make it a static IP address - to better
organize my home's IP addresses.

Also, does anybody know the limit of LEDs that the ESP8266 can handle?

Thank you SO MUCH!!!

mariusmotea, 2019-01-23:
Few tips:

- if you perform a light search, the ip's of the lights will be updated in configuration based on mac address match
- dhcp reservation works for me perfectly.

charlescbullock, 2019-01-24:
DHCP reservation on the router?
Based on the Mac Address of the ESP8266?

I'll give it a shot!

Thank you again so much!!

Chris Bullock

Technical Director - Cokesbury UMC

Audio, Lighting, & Video Engineer
charlescbullock@gmail.com
(770)880-6858
[image: https://www.linkedin.com/in/charles-bullock-4a64a124]
<https://www.linkedin.com/in/charles-bullock-4a64a124>
<https://www.facebook.com/C.C.Bullock>
<https://www.instagram.com/thechrisbullock/>
<https://twitter.com/thechrisbullock>
<https://plus.google.com/107127380563416500310>

Mevel, 2019-01-24:
The number of LEDs can be high >100 but you have to make sure your power supply can handle the load.

charlescbullock, 2019-01-29:
Reservation through the router seems to work!

In terms of LED count, I JUST read of someone driving over 1300 pixels with an ESP8266..
I have a 5v 60amp power supply.. Should be able to handle around 900 LED's with the power supply alone.

I know I have controlled one full roll of 150 LEDs without failure.

I have not tried to shoot for the stars yet, but I'm going to try to use up to 300... 

Has anyone had experience with more than 150?

charlescbullock, 2019-01-29:
I'm going to ask this in a general "issue" under the Lights page.. This might make it easier for someone else to find, if looking for info.

Thank you all!!!!!!

closed 2019-01-29 by charlescbullock