diyhue/Lights #7: Pin Output for Relay Control

ryancasler, 2019-01-27:
I'm hoping you'll be able to point me in the right direction.  What I am trying to do is to use one of the digital pins to control a simple relay to control the power to my WS2812 LED strips.  I have the strips all set up and working perfectly with the sketch in the Arduino folder but I was hoping that I could control the power to the strips when they are off.  Since they are individuallly addressable, they still consume power when they are not on.  And I travel for work so I am not home for long stretches of time.  So, why power the LEDs if there's no chance of them having to turn on any time soon?  I tried using a wifi switch to turn power off to the whole setup but the boot sequence takes a little while and is a bit of a pain.  So, I was hoping that you could point me to some variable within the Arduino sketch that I could use to determine if the light was on or off.  Then I could use that to write a digital pin high or low depending on that value.  I tried a few different things but with no luck.  Any assistance you could give would be greatly appreciated.  Don't need help with the code, just some variable I could compare to or check the truth of would be a great first step.

Thanks!  And thanks for this amazing project!  Has saved me a boatload of cash and really worked well for me.  Please, keep up the great work!!!  And if you ever need beta testers, I've got a bunch of different ESP boards and LED strips/sticks.  I'd be happy to help in any way I can.

Mevel, 2019-01-27:
I tried one idea i had in mind but found a small Problem. I think, if you power off the WS2812B they need to be reinitialised upon powerup. Not sure.

Anyway this is my Idea i came up with (quick, assuming active LOW):

`#define relay0Pin 14`

in `void Setup()` `pinMode(relay0Pin,OUTPUT);
  digitalWrite(relay0Pin, LOW);` before Strip is initialised.

In `void lightEngine()`, check status of first or the only Lamp in the strip `light_state[0]`. According to the value, enable Relay.

If you need to reinitialise the strip you have to perform a reset of ESP.

I am unaware of the boot behaviour of those pins so this might work or not. Also i encounterd Initialising pins before other librarys might not work.

You can try this.
[Pastebin Source Code](https://pastebin.com/3kvXBXMe)

Mevel, 2019-01-27:
On a Sidenote:
1x WS2812 OFF  =  1mA
Assume 100 LEDS its 100mA 
x24h
2400mAh x 5V = 12Wh/d
x365d = 4380Wh/yr
x0,3€/kWh = 1,314€/yr

So you might just want to let it on anyway and save the stresss =)

mariusmotea, 2019-01-27:
You need to set one free gpio pin to be high or low when there is a get request that contain the 'on' parameter. I already have some extra functions performed on this paramerer. From my point of view will not worth because the esp device may consume more than all the leds in idle.

ryancasler, 2019-01-27:
> esp device may consume more than all the leds in idle.

Yes, but the esp is going to be on anyway.  So, that's not really a factor.  

> So you might just want to let it on anyway and save the stresss =)

Yes, I know.  I did the math a while ago too and came up with an even smaller number here in the US.  But I do have about 150 LEDs currently and looking at adding about 300 more for under-cabinet lighting.  But i'm not really stressing over it so much as just curious if it's possible.  That's all.

mariusmotea, 2019-01-27:
It is possible and very simple i believe. You will lose the dimming effect when you turn the leds off...if you don't create a delay circuit.

ryancasler, 2019-02-02:
I'm sorry, I'm not following you.  It would only turn the relay off when the lights were off and as soon as they were turned on it would turn the relay on.  The whole point is that the relay is on only when the lights are on and only off when the lights are off...automatically.  I'm trying to control the power to the LEDs when they are off anyway to prevent the LEDs from consuming any electricity when they are off.  Since they are neopixels and not analog LEDs, they draw power, even when they are off.

mariusmotea, 2019-02-02:
I mean that when off command is received in reality the leds are not directly turned off, but a dimming effect will start to be applied every 6ms until the leds are completly off. This take in general about 0.4 seconds, your releay will cut the power in 0.1 seconds.

mariusmotea, 2019-02-02:
i check now the code, i believe there is one possible way to obtain that:

```
  if (in_transition) {
    delay(6);
    in_transition = false;
```

in this if statement add another one:
```
if (!light_state[light] && current_rgb[light][0] == 0 && current_rgb[light][1] == 0 && current_rgb[light][2] == 0) {
      digitalWrite(relayPin, LOW);
    }
```
!light_state[light] = light was set to off
current_rgb[light][0] = red arrive to point 0
current_rgb[light][0] = green arrive to point 0
current_rgb[light][0] = blue arrive to point 0

remember to set the relay pin as output in `setup()`

to turn the relay on where you find `light_state[light] = true;` add also `digitalWrite(relayPin, HIGH);`

Remain one issue, the relay will change when you play with single virtual light on the strip. To avoid this you need to write more code.

ryancasler, 2019-02-10:
I'm sorry, I'm more confused now.  If 
light_state[light] = true
Should be used to turn the relay on, shouldn't 
light_state[light] = false
be used to turn the relay off?  I think you are making it more complicated then it has to be.

mariusmotea, 2019-02-10:
It will turn of the light without to play the fade effect.

closed 2019-02-19 by stale[bot]