diyhue/Lights #8: ESP8266: WS2812B - Maximum Control Quantity

charlescbullock, 2019-01-29:
Hello all!

I have successfully controlled a roll of 150 WS2812B LEDs.
I am using a 5v 60amp power supply, and should be able to power 900 LED's off of one supply..

I'm not necessarily planning on doing just that.
Maybe 400.

I wanted to see if anyone knew the maximum number of LEDs one can control, using a single ESP8266 microcontroller. 

I JUST read of an individual controlling 1300 pixels..
Not sure how it was wired.. 

Thank you all!

Nikfinn99, 2019-01-29:
The maximum number is technically not limited but at some point you will run out of memory space on the esp. The more LED's you have the longer it will take to update all of them, as the bus is clocked at 800kHz. with 300 LED's you will not achieve 60fps but animations will still be smooth.

Nikfinn99, 2019-01-29:
I've used up to 300 LED's in the past from one controller without any problems, but the power supply becomes an issue there if all leds are at may brightness. The WS2812b Strip itself can not carry enough current for 300 leds so the last leds will not reach full brightness.

mariusmotea, 2019-01-29:
@charlescbullock why you want to achieve this result with only one ESP?

charlescbullock, 2019-01-29:
I'm looking to install several ESPs for this project, and I'm looking to
span about 5 meters with each.

I can use the 150 LED 5m strips that I have, or use 300 LED 5m strips.

I was contemplating ways to use about 15 meters each as well.. And what
really made sense in the end.

I'm installing accent lighting around the house. Haha!

charlescbullock, 2019-02-01:
Thank you all, again for the help!

I'm going to close this I suppose!

closed 2019-02-01 by charlescbullock