diyhue/Lights #9: Color conversion for WS2812 (xy to rgb)

Paalap, 2019-01-29:
I was wondering if there was an intention for the current setup in the sketch - I've changed this for me, but maybe somebody is also looking for similar issue...

What I'm talking about: set a color to 0 saturation and full brightnes in the hue app and you'll see (almost) white but not with the full bright.

Nikfinn99, 2019-01-29:
I've noticed this issue before and solved it in my sketch.
After // Apply gamma correction // and the corresponding three lines of code you should replace the if statements with:

```
  float maxv = 0;// calc the maximum value of r g and b
  if (r > maxv) maxv = r;
  if (g > maxv) maxv = g;
  if (b > maxv) maxv = b;

  if (maxv > 0) {// only if maximum value is greater than zero, otherwise there would be division by zero
    r /= maxv;   // scale to maximum so the brightest light is always 1.0
    g /= maxv;
    b /= maxv;
  }
```
This will scale the values always up to the maximum 1 so lights can achieve full brightness.
the brightness is added later with `(int) (r * optimal_bri)` so they will display correctly.

Paalap, 2019-01-29:
This is my solution (commented lines are the old lines):

// Apply gamma correction
//r = r <= 0.04045f ? r / 12.92f : pow((r + 0.055f) / (1.0f + 0.055f), 2.4f);
r = r <= 0.0031308 ? 12.92f * r : (1.0f + 0.055f) * pow(r, (1.0f / 2.4f)) - 0.055f;
//g = g <= 0.04045f ? g / 12.92f : pow((g + 0.055f) / (1.0f + 0.055f), 2.4f);
g = g <= 0.0031308 ? 12.92f * g : (1.0f + 0.055f) * pow(g, (1.0f / 2.4f)) - 0.055f;
//b = b <= 0.04045f ? b / 12.92f : pow((b + 0.055f) / (1.0f + 0.055f), 2.4f);
b = b <= 0.0031308 ? 12.92f * b : (1.0f + 0.055f) * pow(b, (1.0f / 2.4f)) - 0.055f; `

Nikfinn99, 2019-01-29:
Have you tested your solution? 
I created a pull request with my solution that i've already tried and which looks perfect.

Paalap, 2019-01-29:
for me my solution is working fine, I wanted to undestand if there was an idea to define it in the existing way :)

Nikfinn99, 2019-01-29:
Ok i haven't tested different gamma curves like those you have proposed.
No idea what source code @mariusmotea is relying on for gamma correction as there are many different conversion options from xy to rgb depending on which exact color space you are using.
I think he used the guidelines from phillips they have provided for implementing color conversion.

Paalap, 2019-01-29:
I was actually lazy and used the curves from HueEmulator3.py , these are working good from my point of view ;)

Nikfinn99, 2019-01-29:
I've plottet the different gamma curves:
[image]
green is the original curve by @mariusmotea and red is your proposed curve.

both curves should intersect the point [1,1] but the green curve is too flat.

Your red curve is the inverse gamma curve so it is correct, as should be controlled similar to the green curve, to achieve linear perceived brightness.

Nikfinn99, 2019-01-29:
So i've made a mistake when entering the original green function.
this is the correct plot:
[image]
So the currently used gamma curve is absolutely correct, and the brightness scaling is probably something related to the matrix transforms, which is fixed by my proposed solution.

Nikfinn99, 2019-01-29:
> I was actually lazy and used the curves from HueEmulator3.py , these are working good from my point of view ;)

Thats exaclty why you used the inverse function ;)
when converting in the other direction, the inverse function is used, but applying the same function twice as you currently do will create the wrong colors.

Paalap, 2019-01-29:
sorry it was def convert_xy from color.py to convert xy to rgb, I would say the direction is ok :)

mariusmotea, 2019-01-30:
Just some info. The conversion function from color.py is used just for MiLight and YeeLight bulbs that don't understand xy values. For esp lights, the conversion is inside the arduino sketch. Every color bulb need to have his own conversion formula based on leds color reproduction, what i added is the standard formula from hue api that work in general ok for most of the lights.

cheesemarathon, 2019-02-01:
In that case, should some documentation be created that describes how to "tune" the sketch to improve colour output for your lights?

PR #10 (2019-02-03): correct brightness when converting from xy to rgb 
When selecting white light in the hue app from xy colo space, pure white will only reach 1/3 of the maximum brighness. Brightness ist supposed to be added separately.

```
  float maxv = 0;// calc the maximum value of r g and b
  if (r > maxv) maxv = r;
  if (g > maxv) maxv = g;
  if (b > maxv) maxv = b;

  if (maxv > 0) {// only if maximum value is greater than zero, otherwise there would be division by zero
    r /= maxv;   // scale to maximum so the brightest light is always 1.0
    g /= maxv;
    b /= maxv;
  }
```
this replaced the previo [... cut]
closed 2019-02-03 by mariusmotea