diyhue/Lights #13: Generic onoff arduino

sommo, 2019-02-01:
Uploaded to a wemos d1 mini, comment a 4relay board, added to the diyhue bridge but the realtime status is a mess.. when i try to turn on One of the four output it succedd but after few second all the 4 “bulbs “ change state no way to know the real state of them

Mevel, 2019-02-01:
You can modify `Line 45` in the `Setup()` function to set the initial state upon powerup.

`for (uint8_t ch = 0; ch < devicesCount; ch++) {
    pinMode(devicesPins[ch], OUTPUT);
digitalWrite(devicesPins[ch],LOW;
}`

Remark: depending on your Relay Board (active LOW or active HIGH) you need to change the powerstate.

sommo, 2019-02-01:
Thank you Mavel, i Willy flash it again with this Line modified.
What about the major issue? Thank you!

Inviato da iPhone

mariusmotea, 2019-02-01:
I will try to replicate this error with an empty wemos. Looking to the code i don't find the problem.

sommo, 2019-02-02:
Thank you!
Let me know.
I flashed a new wemos d1 mini, all process (flash/wifimanager/add the light via hueapp) went good but it doesnt works like should be.
It seems something wrong on getting its real time status. Once turning on 1 of 4 bulb, all seems ok but after few seconds (about 4 seconds) every bulbs turn to on. this happen in diyhuebridge webpage while trying to control one bulb too.
Let me know Also if you need log.
The project is amazing!
Matteo

Inviato da iPhone

mariusmotea, 2019-02-02:
i will need the debug log because i was not able to replicate this. I believe there is an error in the function that fetch the lights status every 15 seconds when hue app is active, so i want to see the error

sommo, 2019-02-02:
Can you point me on how to get the log?

Inviato da iPhone

sommo, 2019-02-02:
Here https://diyhue.readthedocs.io/en/latest/AddFuncts/debug.html?highlight=Debug

sommo, 2019-02-02:
Really dont know what happend but once tried the debug mode, all works like a charm, so i stopped the debug and launch the systemctl and magically work like it should do.
Only difference i put a dhcp reservation but i cant imagine that was this the problem cause if it fails to connect it will return to wifimanager mode (right?).

Thank you for help me in troubleshooting.. i keep testing anyway.

A question, could i add the hue emulator bridge into homekit ecosystem? If so, How could i do?

Inviato da iPhone

sommo, 2019-02-02:
Maybe I found a way, throught homebridge and the homebridge-hue plugin. I will let you know so to share with the community

mariusmotea, 2019-02-02:
Please share your findings with homebridge-hue, lot of people ask about this.

sommo, 2019-02-03:
I successfully added your diyhue bridge and light to the HomeKit !
I will share soon how to do it step by step

sommo, 2019-02-04:
> You can modify `Line 45` in the `Setup()` function to set the initial state upon powerup.
>
> `for (uint8_t ch = 0; ch < devicesCount; ch++) { pinMode(devicesPins[ch], OUTPUT); [...]

I tried to modify the sketch to match my relayboard, active high, but i recieve a compile error with your string modified:

Generic_ON_OFF_device_HUE_relayboardinverted:46: error: expected ')' before ';' token

     pinMode(devicesPins[ch], OUTPUT); digitalWrite(devicesPins[ch], LOW;

                                                                        ^

i really don't know how to modify all the sketch to make it works with my relay board (need to be the opposite of how is build the original sketch.
i need to change all the LOW string to HIGH and viceversa?
Thank you!
p.s. what about powerstate recovery (last features of hue), how does it works with this project? does the rpi/arduino store somewhere the light status and set the it on power on?

mariusmotea, 2019-02-04:
Thanks, this will go in the wiki

sommo, 2019-02-04:
thanks to you for the project, glad to contribute!
let me know for what asked before (relay board inverted), so i could test it and maybe could be useful to publish a generic onoff sketch (for relay board inverted-active high/low).
i also bought a strip led rgb (3 pin), to test with your sketch :) it should arrive tomorrow..

cheesemarathon, 2019-02-04:
Added home kit instructions [here](https://diyhue.readthedocs.io/en/latest/AddFuncts/homekit.html). Would you mind checking them to make sure I formatted it all correctly!?

Mevel, 2019-02-05:
@Mevel there is a typo: missing a `) `

You can join our slack group for help

Https://joinslack.diyhue.org

You can contact me there for live debugging and we will post the result here.

sommo, 2019-02-05:
i joined the slack group and asked for that (sketch arduino generico onoff for relay board active high)

sommo, 2019-02-05:
> 
> 
> there is a typo: missing a `) `
> 
> You can join our slack group for help
> 
> [Https://joinslack.diyhue.org](Https://joinslack.diyhue.org)
> 
> You can contact me there for live debugging and we will post the result here.

so only changed this will make the sketch working for a relayboard active high?

void setup() {
  EEPROM.begin(512);

  for (uint8_t ch = 0; ch < devicesCount; ch++) {
    pinMode(devicesPins[ch], OUTPUT); digitalWrite(devicesPins[ch], LOW**)**;
  }

closed 2019-02-14 by stale[bot]