diyhue/Lights #14: WS2812 Hue Ring - Hue App - Unstable Connection in Official Hue App

charlescbullock, 2019-02-01:
Hey guys!

I was looking to create customized groupings of lights, using WS2812B lights.

I was recommended to use the WS12B Hue Ring sketch.

Did this, and the groupings work!
Woot!!

However, I am experiencing an unstable connection in the Official Hue App.
I am using the sketch, as is - with no modifications.

The "lights" are constantly going between "Connected" and "Unable to Connect" in the app, when interfacing with them.
After I stop touching the screen, for about 2 seconds, the Room and Lights turn to the "Off Position" in app.
However, the lights themselves stay on - in their last used setting. 
The only means of control, I can use are the **Room On/Off switch**, and the **Room Scene Selector**. 
Other than that, I am unable to control values for the "lights" individually.

I can edit scenes, then launch said scenes to control individual values for lights.
But there is no direct control of the lights individually, other than scenes. 

Is this the intention of the WS2812B Hue Ring sketch?

Thanks for your help guys!
I REALLY appreciate everything you have done here!

Chris

ryancasler, 2019-02-02:
Does your ESP reboot?  You can tell if that's happening by the onboard led flashing.  If that's happening then their might be a problem somewhere else that is causing the board to reboot.

charlescbullock, 2019-02-02:
I will try to look into it tonight if I have time!
I'm out of town right now, but will try to get the debug information up as
well as check on the potential reboot!

Thank you guys!

mheijnemans01, 2019-02-03:
I can confirm issues with this setup. In my case, the resets only occur when connected to the leds  through a breakout board. Seems to me the leds are causing a voltage drop, causing the ESP to reset. Can you try connecting the board just with usb, and see if it is stable then? If so, it might be something with the cables or power supply.

mheijnemans01, 2019-02-04:
Allright, i just tested a couple different configurations to reproduce the issue, and i found 2.

Issue 1: Not working with a (cheap) breakout board.
When the led power (5v / 12v) is drawn from the breakout board, the ESP seems unstable while controlling. 
However, when you disconnect the power leads OR the data lead to the ledstrips, it is fully stable for me.

Issue 2: Not working with esp8266 board library > 2.4.2.
When i updated my libraries yesterday, every ESP i programmed rebooted after simple wifi requests.
I'm not sure what the exact issue is here, but downgrading / upgrading the esp8266 board library to 2.4.2 seems to resolve the issue.

charlescbullock, 2019-02-05:
Alright - sorry it has taken me a few days to reply here!!

So - I have tried a few things.

The ESP does not seem to be rebooting when I see the "Connected" and "Unable to connect" lights pop up. They are pretty quick to fire back and forth. Not even a full second between going from "Connected," to "Unable to Connect" and back to "Connected." It's so fast, that I have a hard time reading, what I think says, "Unable to Connect." This information only appears when I'm trying to modify information.

Case 1: I will start by saying, the units work GREAT!!! over ip control from the IP of the light iteself.
If I go to the IP of the Hue Bridge, and try to use the controller you guys have made there - it does fail to connect. Could there be an issue with the bridge configuration? 

I have tried restarting it - which I believe checks for and installs updates to the diyHue Bridge Emulator upon restart.. Right?

Case 2: I have tried a new (cheap) breakout board, and reproduced more errors! haha! 
This was before the last message. I will be returning those units.. Do you guys have any recommendations in terms of ESP8266 units to get in replacement of these?

Case 3: I tried to revert back to the WS2812BHueStrip.ino configuration (that was previously working), and reproduced the same "Connection Issue" with the ESP8266 that had worked with this configuration in the past. However, this was not an issue prior to trying the Ring configuration.. Which is where I began to wonder if I'm doing something wrong when I'm flashing the unit? Is there a "preferred method" for preparing the unit to write to flash? I have been using the "Hold Flash and then hit Reset" method.. Or vice-versa. I can't remember which one it is. I just make sure the LED blinks when I release the second button.
Anyways, I do this after I compile the sketch in the Arduino IDE, but immediately after I see the progress bar at 100%, and the first 3 lines of white text appear at the bottom - stating memory limits and whatnot.

Case 4: I have tried to move the ESP around the house, closer and further from the router. No improved results.

Case 5: I have checked the version in by board library, and I am using 2.4.2.

Case 6: I have tried pulling the ESP out of connection with the WS2812B entirely. No better results.

Case 7: I have tried pulling power from WS2812B with no better results. 
(I am using an external 5v 60A power supply for the strips. I share the ground connection to the strip between the power supply and ESP8266. I also put a 16v (I believe) 1000µF capacitor between the 5v and ground lines going to each strip. Oddly, I tried a 330 ohm resistor in the RX line going to the strip, and I didn't get ANY information to it.. I JUST got these resistors in, and I'm beginning to think they sold me something MUCH GREATER than 330.. I'll check and if it's higher than 330, I'll post again.

Thanks again!
I have to run to work, but I will try to get debug information when I get back in a little while!

charlescbullock, 2019-02-05:
ALRIGHT.
I MIGHT HAVE FIGURED THIS ONE OUT.

So - I had been using the same ESP in different configurations - testing the lights and whatnot.. 

When I ran it in debug mode, I got this: 

`Debug Enabled
2019-02-05 14:39:30,502 - root - INFO - b827ebac7311
2019-02-05 14:39:30,503 - root - INFO - IP range for light discovery: 0-255
2019-02-05 14:39:30,505 - root - INFO - Config loaded
2019-02-05 14:39:30,511 - root - INFO - sync with lights
2019-02-05 14:39:30,555 - root - INFO - Starting httpd...
2019-02-05 14:39:30,561 - root - INFO - Starting ssl httpd...
2019-02-05 14:39:30,639 - root - ERROR - light 2 is unreachable
Traceback (most recent call last):
  File "./HueEmulator3.py", line 910, in syncWithLights
    updateGroupStats(light)
  File "./HueEmulator3.py", line 781, in updateGroupStats
    if bridge_config["lights"][group_light]["state"]["on"]:
KeyError: '4'
2019-02-05 14:39:30,661 - root - ERROR - light 3 is unreachable
Traceback (most recent call last):
  File "./HueEmulator3.py", line 910, in syncWithLights
    updateGroupStats(light)
  File "./HueEmulator3.py", line 781, in updateGroupStats
    if bridge_config["lights"][group_light]["state"]["on"]:
KeyError: '4'
2019-02-05 14:39:30,691 - root - ERROR - light 1 is unreachable
Traceback (most recent call last):
  File "./HueEmulator3.py", line 910, in syncWithLights
    updateGroupStats(light)
  File "./HueEmulator3.py", line 781, in updateGroupStats
    if bridge_config["lights"][group_light]["state"]["on"]:
KeyError: '4'
^C2019-02-05 14:40:16,142 - root - INFO - config saved
Traceback (most recent call last):
  File "./HueEmulator3.py", line 1873, in <module>
    sleep(10)
KeyboardInterrupt
^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
`

Lights 1, 2, and 3 were all iterations of the same ESP8266 unit..
After uninstalling, and reinstalling the Hue-Emulator - I am no longer receiving the connection issue. 

Is there a way to delete fixtures in the Hue-Emulator easily, without having to uninstall/reinstall it?

Thanks again guys!

cheesemarathon, 2019-02-07:
Were you not just able to delete the bulbs in the Hue app?

charlescbullock, 2019-02-07:
I was, but it seemed to have issues with reusing the same ESP modules after
removing them in the app alone.

The issue reoccurred after reinstalling the emulator again in the future..
So that theory is wrong.
But I also noticed that some of the lights I hadn't deleted in the app,
before reinstalling the Hue-Emulator, had reappeared immediately upon
connecting my app to the emulator.
Note, I also uninstalled the Hue App, and reinstalled it - and this
occurred.

So - I ended up going to the app settings in my Android Settings Menu, and
cleared All App Data, and the App Cache Data..
The lights were no longer there..

So maybe that's the trick?

Remove the lights from the bridge in the app, and then clear all app data?

Thanks again guys, for everything.
I can't wait to show you some pictures!
The house looks like a bomb went off in our kitchen.. and at my workbench..
haha!

closed 2019-02-16 by stale[bot]