diyhue/Lights #15: WS2812BHueRing Light Order

charlescbullock, 2019-02-05:
Hey guys!
Sorry for another post. I've had several questions recently, and you have no idea how much I appreciate the help!
What you are doing is AWESOME! 
I've produced some beautiful arrangements with it! I'm trying to install them in my kitchen now!

I'm just trying to tie down the Ring sketch's light order by pixel count.

With the Ring sketch, how does one order the pixel(LED) groups? 

I have a 39 pixel strip, and I'm trying to program it into equal(ish) parts for use with scenes in my kitchen.

I suppose a symmetrical way to order the sections of this LED Strip would be 5 sections with the following order:

Under Cabinet Light 1: 7 LEDs
Under Cabinet Light 2: 8 LEDs
Under Cabinet Light 3: 9 LEDs
Under Cabinet Light 4: 8 LEDs
Under Cabinet Light 5: 7 LEDs

Thank you all!!

mariusmotea, 2019-02-06:
Hi,

`uint16_t pixelStart[lightsCount + 1] = {0, 1, 21, pixelCount};`

the rule is this. the array has one extra element in the end and this is `pixelCount` and it start with 0.

in your setup must be:

`uint16_t pixelStart[lightsCount + 1] = {0, 7, 15, 24, 32,  pixelCount};`

charlescbullock, 2019-02-07:
Alright, so here is what I have:
"
#define light_name "Kitchen Kickplate South"  //default light name
#define lightsCount 5
#define pixelCount 39
uint16_t pixelStart[lightsCount + 1] = {0, 7, 8, 9, 8, pixelCount};
"
I did this thinking it would give me 5 sections within the 39 pixels.
I thought it would give me:
Light 1: 7 LEDs
Light 2: 8 LEDs
Light 3: 9 LEDs
Light 4: 8 LEDs
Light 5: 7 LEDs

So, when I ran it, this is what I got control of:

Light 1: 7 LEDs
Light 2: 1 LED
Light 3: 1 LED (but Light 5 takes over this LED when modified)
Light 4: No Control
Light 5: 31 LEDs (including Light 3's LED)

Is this operating properly?

Does the order of pixel counts have to be progressing in size, from
smallest to largest?
I took the original sketch, WS2812BHueRing and tried:
"
#define light_name "WS2812 Hue Ring"  //default light name
#define lightsCount 3
#define pixelCount 61
uint16_t pixelStart[lightsCount + 1] = {0,* 21, 1, pixelCount*};
"

Light 1: 1 LED
Light 2: 21 LEDs
Light 3: 39 LEDs

I might just use the WS2812BHueStrip.ino file, and put one or two extra
LEDs in the code - but not on the strip to make it fairly even..
So in this case, I'd just state that I'm using 40 LEDs in the Arduino file,
and really only use 39.. Shouldn't be an issue.. Hopefully!

"
#define light_name "Kitchen Kickplate South" //default light name
#define lightsCount 5
#define pixelCount 40
#define transitionLeds 4 // must be even number
"

Thank you again!

mariusmotea, 2019-02-10:
Hi, you don't declare the lights as i show you. you put `{0, 7, 8, 9, 8, pixelCount};` instead of `{0, 7, 15, 24, 32, pixelCount}`. The logic is not that basic, you must increment the number of the leds with previews light, not to specify directly the number of leds per light.

ex: `{0, 7, 15` that 15 represent 7 + 8 leds.

charlescbullock, 2019-02-13:
That makes sense!

Sorry about that.

I haven't had a chance to give it a shot yet - but I will hopefully
tomorrow or the following day!

Thanks again!

closed 2019-02-22 by stale[bot]