diyhue/Lights #16: Arduino/Generic_Dimmable_Light ESP8266 crashes using only one pin [Bug]

Tederean, 2019-02-10:
Hi folks,

i am fairly new in using github active, so please excuse some mistakes.

Lights/Arduino/Generic_Dimmable_Light/Generic_Dimmable_Light.ino sketch triggers an illegal array access in line 166 and 168 when using only one pin (one fake hue device):

`if (! light_state[0])  {
  // Show that we are connected
  analogWrite(pins[1], 100);
  delay(500);
  analogWrite(pins[1], 0);
}`

I can replicate this problem, using two pins no erros occuring, using only one pin triggers a crash after pairing with diyHue, replacing pins[1] with pins[0] in line 166 & 168 fixes the problem.

Mevel, 2019-02-10:
you are right!
this sketch assumes you have a RGBW Setup. and "Show we are connected" wants to blink the green LED.

will be fixed! Thank you!

Mevel, 2019-02-10:
commit 98b2a4abadb7a74f33c133098f90fef5a5fea515

closed 2019-02-10 by Mevel