diyhue/Lights #18: Cannot build PlatformIO Generic_RGB_Light

vhsw, 2019-02-16:
Hello!

I can't build PlatformIO Generic_RGB_Light
```
Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <WifiManager> 0.14
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266WiFi> 1.0
Compiling .pioenvs/nodemcuv2/src/Generic_RGB_Light.cpp.o
In file included from src/Generic_RGB_Light.cpp:10:0:
src/pwm.c: In function 'uint8_t _pwm_phases_prep(pwm_phase*)':
src/pwm.c:212:11: warning: unused variable 'off_mask' [-Wunused-variable]
uint16_t off_mask = 0;
^
src/Generic_RGB_Light.cpp: In function 'void convert_hue()':
src/Generic_RGB_Light.cpp:58:17: warning: right operand of comma operator has no effect [-Wunused-value]
hh /= 11850, 0;
^
src/Generic_RGB_Light.cpp: At global scope:
src/Generic_RGB_Light.cpp:149:3: error: 'r' does not name a type
r = r <= 0.04045f ? r / 12.92f : pow((r + 0.055f) / (1.0f + 0.055f), 2.4f);
^
src/Generic_RGB_Light.cpp:150:3: error: 'g' does not name a type
g = g <= 0.04045f ? g / 12.92f : pow((g + 0.055f) / (1.0f + 0.055f), 2.4f);
^
src/Generic_RGB_Light.cpp:151:3: error: 'b' does not name a type
b = b <= 0.04045f ? b / 12.92f : pow((b + 0.055f) / (1.0f + 0.055f), 2.4f);
^
src/Generic_RGB_Light.cpp:153:3: error: expected unqualified-id before 'if'
if (r > b && r > g) {
^
*** [.pioenvs/nodemcuv2/src/Generic_RGB_Light.cpp.o] Error 1
======================================================= [ERROR] Took 1.39 seconds =======================================================

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 1.0
|-- <ArduinoOTA> 1.0
[... 618 lines cut ...]
.pioenvs/esp12/src/pwm.c.o: In function `pwm_init':
pwm.c:(.irom0.text+0x3c4): multiple definition of `pwm_init'
.pioenvs/esp12/src/Generic_RGB_Light.cpp.o:Generic_RGB_Light.cpp:(.irom0.text+0x3c4): first defined here
.pioenvs/esp12/src/pwm.c.o: In function `pwm_get_period':
pwm.c:(.irom0.text+0x4ec): multiple definition of `pwm_get_period'
.pioenvs/esp12/src/Generic_RGB_Light.cpp.o:Generic_RGB_Light.cpp:(.irom0.text+0x4ec): first defined here
.pioenvs/esp12/src/pwm.c.o: In function `get_pwm_version':
pwm.c:(.irom0.text+0x4f4): multiple definition of `get_pwm_version'
.pioenvs/esp12/src/Generic_RGB_Light.cpp.o:Generic_RGB_Light.cpp:(.irom0.text+0x4f4): first defined here
.pioenvs/esp12/src/pwm.c.o: In function `set_pwm_debug_en':
pwm.c:(.irom0.text+0x4f8): multiple definition of `set_pwm_debug_en'
.pioenvs/esp12/src/Generic_RGB_Light.cpp.o:Generic_RGB_Light.cpp:(.irom0.text+0x4f8): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp12/firmware.elf] Error 1
====================================================== [ERROR] Took 12.22 seconds ======================================================

=============================================================== [SUMMARY] ===============================================================
Environment nodemcuv2   [ERROR]
Environment d1_mini     [ERROR]
Environment esp12       [ERROR]
====================================================== [ERROR] Took 35.89 seconds ======================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

mariusmotea, 2019-02-16:
With Arduino is working?

mariusmotea, 2019-02-16:
Ok, i know what is the problem, the new ESP8266 library is not compatible with the PWM library included. Try to use the Arduino sketch because is up to date. It must work also with PlatformIO.

closed 2019-02-16 by vhsw