diyhue/Lights #19: SK6812HueStrip not detected

tibi03, 2019-02-17:
Hello,

i have some problems with SK6812HueStrip , when i flash it no lights are detected by the emulator, if i flash on the same nodemcu board the WS2812BHueStrip firmware lights are detected. 

i am compiling with arduino .

thank you

mariusmotea, 2019-02-17:
Not sure what is the problem, but stay tunned, i will upgarde it to new version like WS2812.

tibi03, 2019-02-18:
hello, i just tested the last version WS2812BHueStrip and it is not stable even if i use only 60 leds.( i have a 144 strip ) .

mariusmotea, 2019-02-18:
What ESP board are you using? I made tests with ESP12E and Wemos D1 mini. Absolutely no stability issues. One important thing that can cause errors is to not select a number of lights that divide the number of leds (ex: 100 or 101 leds with 3 lights).

tibi03, 2019-02-18:
i am using nodemcu board. i did a config with 4 lights 60 leds 

uint8_t lightsCount = 4;
uint16_t pixelCount = 60, lightLedsCount;
uint8_t transitionLeds = 4; // must be even number

mariusmotea, 2019-02-18:
I made an initial SK6812 version but is not tested. The Arduino sketch is locate in develop branch here:

https://github.com/diyhue/Lights/blob/develop/Arduino/SK6812HueStrip/SK6812HueStrip.ino

If you have time you may test this and provide feedback.

tibi03, 2019-02-18:
it seems to be a problem when it is compiled using board manager 2.5.0, with 2.4.1 its ok. its stable only some flickers and random colors while doing on/off command  for WS2812BHueStrip.

i will test the SK6812 

thanks

tibi03, 2019-02-18:
for SK6812 : 
Lights-develop\Arduino\SK6812HueStrip\SK6812HueStrip.ino: In function 'void ChangeNeoPixels(uint16_t)':

SK6812HueStrip:571:9: error: cannot convert 'NeoPixelBus<NeoGrbFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >*' to 'NeoPixelBus<NeoGrbwFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >*' in assignment

   strip = new NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod>(newCount); // and recreate with new count

         ^

exit status 1
cannot convert 'NeoPixelBus<NeoGrbFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >*' to 'NeoPixelBus<NeoGrbwFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >*' in assignment

mariusmotea, 2019-02-18:
I will test in 1 hour

mariusmotea, 2019-02-18:
I fix the compile error, please try again now.

tibi03, 2019-02-18:
Yes, looks fine now also detected by the emulator. Thanks

mariusmotea, 2019-02-18:
Good to heat that, very likely you are the first one that use this sketch. Feel free to write here any issue you may face, if there are no complains in a few days i will push this to master branch

tibi03, 2019-02-18:
from first tests the communication between esp and raspberry it is slow, i am using docker.

mariusmotea, 2019-02-18:
Ping is stable? Normally you must have constant 2-3ms replay from esp. If not there is a networking issue in the middle

tibi03, 2019-02-18:
i just shuffle the scene and after some time the scene were displayed correctly

tibi03, 2019-02-18:
the ping it is constant 2-4ms

tibi03, 2019-02-18:
2 videos while using entertainment:

[20190218_202628.zip](https://github.com/diyhue/Lights/files/2876981/20190218_202628.zip)
[20190218_202900.zip](https://github.com/diyhue/Lights/files/2876987/20190218_202900.zip)

mariusmotea, 2019-02-18:
The videos are very short. I see a flickering in left side, is that unexpected behaviour? Cpu usage on computer may be a reason of delay. Avoid as much as possible other wifi devices. The best experience will be only with the esp device connected to wifi. How many lights did you emulate?

tibi03, 2019-02-18:
Cpu load is low. I have emulated 6 light, 60 leds total. Ping is stable. I dont know what could be the cause. The flicker its unusual. Please can you share the libraries version and board sdk that you are using?

mariusmotea, 2019-02-18:
I will upload binary firmware that you can flash using /update http path of the light

mariusmotea, 2019-02-18:
I found one more issue, `RgbColor` declared instead of `RgbwColor`. You can update the firmware using light GUi `http://{ip}/update` and uploa below file:
https://github.com/diyhue/Lights/raw/master/Arduino/SK6812HueStrip/SK6812HueStrip.bin

tibi03, 2019-02-19:
i think now its using a lot more bandwidth to communicate, i have moved the strip closer to the router and all seems to be better but with hue sync the lights are changing very slow even if i set it to intense , i tested with the music.

mariusmotea, 2019-02-19:
Try more app's that use entertainment protocol because is strange what you are saying, the protocol is UDP so packages if are not delivered are lost. I believe is possible that your computer send frames with delay. Can you check other devices that can communicate with Entertainment? Also if you reduce the number of lights on the strip it may help. Hue Essentials for Android has some entertainment effects.

tibi03, 2019-02-19:
with this https://github.com/mariusmotea/diyHue/blob/master/Lights/Arduino/WS2812BHueStrip/WS2812BHueStrip.ino all it is dynamic as first time i have used this soft with hue sync , music intense 

   with https://github.com/mariusmotea/diyHue/blob/master/Lights/Arduino/SK6812HueStrip/SK6812HueStrip.ino  lights are not detected by emulator

mariusmotea, 2019-02-19:
But you are pointing to my repo that is legacy. Now we push updates only to community repo, this is the one made last time: 
https://github.com/diyhue/Lights/tree/master/Arduino/SK6812HueStrip

tibi03, 2019-02-19:
yes, i know it is an old version but its working as expected 😃

mariusmotea, 2019-02-19:
are you using latest version of HueEmulator?

tibi03, 2019-02-19:
i am using the docker version

mariusmotea, 2019-02-19:
I made a test now on a Ubuntu Vm and detection is working. Not sure what can be the problem in your case.

tibi03, 2019-02-19:
i did a test in ubuntu vm, with old firmware WS2812 it is detected , SK6812 its not

tibi03, 2019-02-20:
it is possible to add an option in emulator portal to manually add lights in case autodetect its not working?

mariusmotea, 2019-02-20:
i prefer to work on improving the detection than to add this future as this also require time.If somebody want to create a new webui for this i have no problem.

tibi03, 2019-02-20:
i dont think it is a problem with the detection since last version of firmware for esp it is detected on both ws2812 and sk6812.

tibi03, 2019-02-20:
i just tested with this https://github.com/mariusmotea/diyHue/blob/master/Lights/Arduino/SK6812HueStrip/SK6812HueStrip.ino but with some changes and it is detected now : 

server.on("/detect", []() {
    //char macString[50] = {0};
    //sprintf(macString,"%02X:%02X:%02X:%02X:%02X:%02X",mac[0],mac[1],mac[2],mac[3],mac[4],mac[5]);
    //server.send(200, "text/plain", "{\"hue\": \"strip\",\"lights\": " + (String)lightsCount + ",\"modelid\": \"LST002\",\"mac\": \"" + String(macString) + "\"}");

    server.send(200, "text/plain", "{\"hue\": \"strip\",\"lights\": " + (String)lightsCount + ",\"name\": \"" light_name "\",\"modelid\": \"LST002\",\"mac\": \"" + String(mac[5], HEX) + ":"  + String(mac[4], HEX) + ":" + String(mac[3], HEX) + ":" + String(mac[2], HEX) + ":" + String(mac[1], HEX) + ":" + String(mac[0], HEX) + "\"}");
  });

👍

mariusmotea, 2019-02-20:
That will work only if you use outdated hue emulator.

tibi03, 2019-02-20:
it is the docker image that i am using now on ubuntu installed yesterday:

`  "Id": "sha256:9e3b692befa80b59269f44daff1a3ebcab88e42040d1c126099a565af24f67f3",
        "RepoTags": [
            "diyhue/core:latest"
        ],
        "RepoDigests": [
            "diyhue/core@sha256:02d8ae682927ef5365ba4675028b2f73b432ebd1da71f981c2236e5eb720b7cd"
        ],
        "Parent": "",
        "Comment": "",
        "Created": "2019-02-19T00:56:04.994398779Z",
        "Container": "d2a77e4e5f14a6495fad445631386d8498da1e4277569beea230ff5fea4bacbd",
`

tibi03, 2019-02-20:
ok, so you did an update also on docker image because now all its fine, tested with both ws2812 and sk6812, dynamic its as expected. 

thank you

mariusmotea, 2019-02-20:
it must be automatically updated. can you try host installation and increase the timeout from 0.02 seconds to 0.1 in HueEmulator3.py file? Is easy to uninstall it after.

sneak-thief, 2019-02-20:
> 
> 
> I found one more issue, `RgbColor` declared instead of `RgbwColor`. You can update the firmware using light GUi `http://{ip}/update` and uploa below file:
> https://github.com/diyhue/Lights/raw/master/Arduino/SK6812HueStrip/SK6812HueStrip.bin

Slightly off-topic: Does OTA still work in addition to this new firmware update method?

mariusmotea, 2019-02-20:
i decomission OTA becase is extra service that consume storage and resources. My plan is to not have the user performing the updates anymore, these must be done with one confirmation click from the hue emulator.

tibi03, 2019-02-21:
installed on host, the same result, lights are lazy

mariusmotea, 2019-02-21:
Can you put an youtube video showing your phone to see when you send the command and when the strip start to change?

tibi03, 2019-02-22:
here is the video with Hue sync:

old fw:
https://youtu.be/xVvERLVd-Sk 

latest fw:
https://youtu.be/-5yXmbG-J_I

while using the app to change color or scene, i dont see a delay.

mariusmotea, 2019-02-22:
I will setup a test env for entertainment.

mariusmotea, 2019-02-26:
I confirm the issue. I still have i wired problem that i don't understand with entertainmentRun variable, when i set this to true, in loop function it remain always false. I fix the delay problem, once i fix also this i will create new PR.

mariusmotea, 2019-02-26:
The issue was with the hue emulator script. I fix it, now i will made a new commit to the strip with small improvement (smooth reverse to previews state once entertainment service is stopped)

closed 2019-02-26 by mariusmotea