diyhue/Lights #21: My9291 Arilux e27 crash loop

rammstone83, 2019-02-27:
Hello, yesterday i used the newest sketch of the my9291 light which worked fine in the past, now its stuck in an wtd reset loop:

 ets Jan  8 2013,rst cause:4, boot mode:(3,5)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

mariusmotea, 2019-02-27:
Hi,

i cannot help you here as i don't have this type of light. Try to wipe the entire flash with esp_tool.py

rammstone83, 2019-02-27:
NP, i allready tried that with the same outcome, i now try to investigate the issue the best i can, maybe i can find a Solution, ill keep you up 2 date about it.

mariusmotea, 2019-02-28:
i recommend to start with a simple color change sketch. Once that works share it with me and i will apply the extra code used by us

rammstone83, 2019-03-04:
Working on it, but have to go on a Business trip till sunday, so i wont be able to test it till next week

closed 2019-03-13 by stale[bot]