diyhue/Lights #23: SPIFFS.begin() fails on Lolin d1 mini V3.1.0

pluypen, 2019-03-02:
Hi,

I'm new here and just bought some Lolin d1 mini's V3.1.0, wanting to connect a ws2812b as an ambilight for my monitor.

When trying to get it all to work, I ran into trouble getting my d1 mini connected to the wifi. Here is what i did:

1. Downloaded diyhue/lights
2. Opened Generic_WS2812_Strip.ino in Arduino IDE
3. Adjusted lightscount to 10
4. Adjusted pixelcount to 80
5. Adjusted transitionleds to 2
Saved the script
6. Compiled the script
7. uploaded the script using Arduine IDE
8. Tried searching for wifinetwork: Hue rgb strip. I couldn't find the network or anything remotely close
9. Went back to Arduino IDE and looked for: void setup()
10. Uncommented: Serial.begin(115200);
11. Uncommented: Serial.println("Failed to mount file system");
12. started serial monitor
13. Saved, compiled and uploaded the script again
14. Got the message in serial monitor: Failed to mount file system
15. Went back to Arduino IDE and commented:
 //  if (!SPIFFS.begin()) {
////    Serial.println("Failed to mount file system");
//    return;
//  }
16. Saved, compiled and uploaded the script again
17. Tried searching for wifinetwork: Hue rgb strip
18. Found the network this time
19. configured the network, went to the ip adress I got from the serial monitor
20. Could control the lights. But they are slow to respond. 
21. Trying to change the transitionleds via the ui, crashed the d1 mini

Have I missed some steps causing SPIFFS.begin() to fail?

mariusmotea, 2019-03-02:
Please avoid any beta library and erase ell flash on first write.

Nikfinn99, 2019-03-02:
Have you compiled the script with spiffs enabled?
[image]
otherwise no memory will be reserved and SPIFFS.begin() will fail.

pluypen, 2019-03-02:
Yes! Enabling SPIFFS did the trick. 
After setting it to 1M SPIFFS, the script ran without issue. I can now change settings via the UI too.
Thank you so much for your comments

closed 2019-03-02 by pluypen