diyhue/Lights #27: can't find wemos d1 mini in app, but web page works (generic ws2812 strip)

sladaar, 2019-03-09:
Sorry to open a new issue but i'm trying to figure out why i can't add a Generic ws2812 strip!
Running the latest Emulator on a pi3 (tried both docker and host installation).

In the app i can see the other lights (Yeelight and tasmota's sonoff), but no Wemos d1 mini flashed with arduino ide.
Sorry for the noobsness, with nmap "wemos ip" it says "host seems down", but using port 80 i can ping it.

Thanks in advance for any kind help!

mariusmotea, 2019-03-10:
Did you join the wemos to your wifi network?

Paalap, 2019-03-10:
Maybe increasing the timeout would solve the issue. For me it was necessary

sladaar, 2019-03-10:
> 
> 
> Did you join the wemos to your wifi network?

thanks for the reply! Yes of course, i was able to control it by the web interface. I tried both dhcp and static ip to be sure it was a router problem.

sladaar, 2019-03-10:
> 
> 
> Maybe increasing the timeout would solve the issue. For me it was necessary

Thanks to you too! Do you mean, increasing in the sketch?

Paalap, 2019-03-10:
No actually in hue emulator for host scanning

mariusmotea, 2019-03-10:
If you have big latency in the network it will not detect it. Search for timeout in HueEmulator3.py file. Not sure what was the value last time, maybe 0.02. Made it 0.1 to be sure.

sladaar, 2019-03-10:
ok gonna check and post the result, many thanks. 
(i ordered other wemos and nodemcus to eliminate the hardware possibility of not finding the esp)

sladaar, 2019-03-10:
So i changed timeout to 0.01 and tried the research but same result, no lights found. 
0.04 and 0.10 either (and yes i wait even after the "no lights found" message, i read the guide!).
Reverted to 0.02 then.

I don't know if it mean something but when i try "nmap -p 80 192.168.1.118" (wemos ip) it says host down (same result without the specified port).
But i can access it with windows10 (from my pc).

Tig3rch3n, 2019-03-17:
I got the same Problem but i use the Generic_SK6812 sketch.
I already tampered with the Timeout but 0.1 still doesn't help

My IPs are HueApp 192.168.178.101
HueEmu                   192168.178.119
Wemos-SK6812       192.168.178.120

debug gets me this
> 192.168.178.101 - - [17/Mar/2019 12:58:16] "GET /api/XXXXX/groups HTTP/1.1" 200 -
>192.168.178.101 - - [17/Mar/2019 12:58:18] "GET /api/XXXXX/lights HTTP/1.1" 200 -
>192.168.178.101 - - [17/Mar/2019 12:58:18] "GET /api/XXXXX/groups HTTP/1.1" 200 -
>192.168.178.101 - - [17/Mar/2019 12:58:20] "GET /api/XXXXX/config HTTP/1.1" 200 -
>192.168.178.101 - - [17/Mar/2019 12:58:21] "POST /api/XXXXX/lights HTTP/1.1" 200 -
>2019-03-17 12:58:21,075 - root - INFO - in post method
>2019-03-17 12:58:21,076 - root - INFO - /api/XXXXX/lights
>2019-03-17 12:58:21,077 - root - INFO - b'{}'
>2019-03-17 12:58:21,658 - root - INFO - sync with lights
>2019-03-17 12:58:27,637 - root - INFO - [
>    "192.168.178.1",
>    "192.168.178.114",
>    "192.168.178.119",
>    "192.168.178.120"
>]
>2019-03-17 12:58:27,764 - root - INFO - {
>    "hue": "strip",
>    "lights": 6,
>    "mac": "cc:50:xx:xx:xx:xx",
>    "modelid": "LST002"
>}
>2019-03-17 12:58:27,765 - root - INFO - ip 192.168.178.120 is unknow device, 'name'
>2019-03-17 12:58:27,772 - root - INFO - registration fail, is the link button pressed?

nmap gives me

>192.168.178.1
>192.168.178.114
>192.168.178.120

and Curl gives me this
>{"hue": "strip","lights": 6,"modelid": "LST002","mac": "cc:50:xx:xx:xx:xx"}

sladaar, 2019-03-17:
@Tig3rch3n thanks for the "up".
Sorry to ask, how to see the debug log?
I read that at least your wemos is recognized in the network (i think mine isn't), which is already something.
But yes, it seems like the same issues I still have. I'm bought and waiting for other Wemos and nodemcu to see if it's hardware related.

I'll post the result.

mariusmotea, 2019-03-17:
@Tig3rch3n please update the firmware of the light.

Tig3rch3n, 2019-03-17:
@sladaar https://diyhue.readthedocs.io/en/latest/AddFuncts/debug.html?highlight=debug
just 2-3 commands
>sudo systemctl stop hue-emulator
>cd /opt/hue-emulator
>sudo ./HueEmulator3.py --debug True

@mariusmotea 
I run the most recent version from the Master branch ... anything newer?
Or have you update something i can't see? Most recent change was 19 days ago

mariusmotea, 2019-03-17:
light firmware not hue emulator script

`{"hue": "strip","lights": 6,"modelid": "LST002","mac": "cc:50:xx:xx:xx:xx"}` this is very old, is missing the name key.

Tig3rch3n, 2019-03-17:
Finally...!
Got my problem

So I accidentally flashed the 'SK6812_WWA' in the first place, after that the reflashing cheeped the saved credentials U_u ... so that explains those ood behaviors :P

Reflash "Generic_SK6812" but set Arduino to delete the Settings
Go to Tools-> Erase Flash-> all flash Contents.

From here it works as expected, connect to the new Wifi, join your house Wifi, connect to the Hue Bridge :)

closed 2019-03-26 by stale[bot]