diyhue/Lights #28: Wemos D1 mini dimmable light not being found in app

Mindstormsnxt, 2019-03-12:
Hello,

I have my bridge emulator, as well as the "dimmable light" firmware up-to-date. Everything relating WS2812 strips firmware works as expected (without issues, also got instantly detected).
However, my esp8266's with the dimmable sketch won't get detected by searching via the app...

If I run ` sudo nmap 10.0.0.62/24 -p80 --open -n | grep report | cut -d ' ' -f5`
The rpi detects the ip-adresses of the lights.

Running `curl 10.0.0.88/detect`  results in:
`{"hue": "bulb","lights": 2,"modelid": "LWB010","mac": "f7:5f:73:eb:f3:84"}`

What am I doing wrong?

Mindstormsnxt, 2019-03-19:
Anyone? Reinstalling the bridge found, again, all lights but the dimmable cct ones inside the app... :/

mariusmotea, 2019-03-22:
Hi,

I believe i now the issue, in 5 minutes i will do a commit.

Marius.

mariusmotea, 2019-03-22:
With commit https://github.com/diyhue/Lights/commit/a3396da18c9c0ad90cb546d8c2fe8ae025377a90 this must be fixed.
Fell free to reopen in case you expect more problems.

Mindstormsnxt, 2019-04-04:
Most lights work by now, however, two wemos d1's still won't show up, although they've been reset, and reconnected to the wifi. The gui of the lights themselves works...

Is it possible that the bridge saved their MAC address and ignores them? A reinstall of the hue emulator software also doesn't work; Searching for new lights works otherwise perfectly fine (with adding new lights and reconnecting the old ones, except for these two...)

cheesemarathon, 2019-04-05:
@mariusmotea looks like there are still issues here

mariusmotea, 2019-04-09:
@Mindstormsnxt to have better visibility we need output. Start the script in debug mode and look if the ip's of the undetected lights are displayed during the light search. To see if the lights remain cached just  search for the mac addresses in the config.json file. The mac address of the light is displayed when you access /detect path of the lights.

Mindstormsnxt, 2019-04-09:
During the search, the ip's od the lights won't show up, but running
`sudo nmap 10.0.0.62/24 -p80 --open -n | grep report | cut -d ' ' -f5`, they all appear.

Searching for the mac, I get:
`{"hue": "bulb","lights": 1,"name": "Kitch_Top_L","modelid": "LWB010","mac": "f7:5f:73:eb:f3:84"}` and
`{"hue": "bulb","lights": 2,"name": "nklab","modelid": "LWB010","mac": "3d:47:a:eb:f3:84"}`

Searching for the mac address, It shows up inside the config.json file, with light nr = 1; 
The second light which isn't working also has this light nr...
One of the Ws2812 lights which is working perfectly fine also has this number defined...?

mariusmotea, 2019-04-09:
Backup the current config, stop the hue emulator and truncate the mac addresses. I believe are not added because are existing lights. If you truncate the mac these will be new lights.

Mindstormsnxt, 2019-04-10:
Hi,

thank you so much for your quick reply! Now, I got every light to show up! I'm still having some delay/performance issues, but I think I can resolve that on my own somehow.

Thanks!

closed 2019-04-10 by Mindstormsnxt

Mindstormsnxt, 2019-04-10:
Just saw that the newly added light (2m next to the perfectly working one) needs about a second longer to turn on / off... Sometimes it just stays lit/dark. Any additional suggestions? :)

mariusmotea, 2019-04-10:
if you ping the light ip from the hue emulator host you have stable ping? It can be because one of your wifi device create noise in the network or poor wifi router. Disable the wifi module from RaspberryPi because it create noise.

Mindstormsnxt, 2019-04-10:
Wow, now I see... It is indeed very unstable... That's strange, we're using Orbi routers and they're covering the entire house. Maybe it's because of this 2.4/5ghz thingy? I'll try changing to another wifi, thanks!
I've already disabled the wifi module, the rpi (3) is conencted via Ethernet.