diyhue/Lights #29: WS2812 Button Control Wiring?

ryancasler, 2019-03-16:
I am trying to figure out how to wire the circuit for the button control in the WS2812 Adjustable sketch.  The sketch says to pull the pin to ground through a 10K resistor but that doesn't actually cause the button to work.  When I look at the sketch, it appears that it is looking for a high signal to read that the button is pushed, so why would I pull it to ground?  Maybe a wiring diagram would be helpful.  For the life of me I can't get this to work with the hardware pins.

mariusmotea, 2019-03-17:
Look this diagram. Difference is that instead of 5v you use 3.3v
https://1sheeld.com/uploads/projects_c5449483-cfd6-420f-a846-08e4cfc8530a.jpg?1412755678285

ryancasler, 2019-03-17:
Okay, I got that far on my own...what I can't figure out is how to get the button to also brighten and dim.  It appears from the sketch that it is supposed to distinguish between a short press and a long press but it doesn't seem to work.  No matter how long or short I press the button, it just turns it on or off.  Is there something I'm missing?

Mevel, 2019-03-18:
Maybe you need to denounce the button.

mariusmotea, 2019-03-22:
on long press the off button must decrease the brightness and on will increase. I did not test the code, but i will explain, maybe you can find some issues:
```
 } else if (use_hardware_switch == true) { // you set in header to use hardware buttons
    if (digitalRead(button1_pin) == HIGH) { //button 1 is pressed
      int i = 0; //counter to see how long was pressed
      while (digitalRead(button1_pin) == HIGH && i < 30) { // while button is pressed increase counter variable every 20ms maximum 30 times (30 x 20 ms = 600ms = 0.6seconds)
        delay(20);
        i++;
      }
      for (int light = 0; light < lightsCount; light++) { //loop to apply the change to all lights
        if (i < 30) { // if counter is smaller than 30, means button was pressed less than 0.6 
          // there was a short press
          light_state[light] = true;
        }
        else {
          // there was a long press, change the brigtness
          bri[light] += 56;
          if (bri[light] > 255) {
            // don't increase the brightness more then maximum value
            bri[light] = 255;
          }
        }
      }
    } else if (digitalRead(button2_pin) == HIGH) { //same like button 1, only that is for button 2
      int i = 0;
      while (digitalRead(button2_pin) == HIGH && i < 30) {
        delay(20);
        i++;
      }
      for (int light = 0; light < lightsCount; light++) {
        if (i < 30) {
          // there was a short press
          light_state[light] = false;
        }
        else {
          // there was a long press
          bri[light] -= 56;
          if (bri[light] < 1) {
            // don't decrease the brightness less than minimum value.
            bri[light] = 1;
          }
        }
      }
    }
```

ryancasler, 2019-03-23:
Yes, I have found that no matter how long or short you press the button it always turns on or off the light and never brightens are dims it.

mariusmotea, 2019-03-23:
can you test following change:

instead of 
```
while (digitalRead(button1_pin) == HIGH && i < 30)
```
put 
```
while (digitalRead(button1_pin) == HIGH && i <= 30)
```

closed 2019-04-01 by stale[bot]