diyhue/Lights #31: Wemos D1 Mini wont broadcast on startup

romanlucas, 2019-03-23:
Hey,
First, thanks for your great work! Enjoying my diyHue since a few months.
Today i did a fresh install of the latest version. Works fine with my yeelights. But i have problems with diy-LED-Strips. Im using SK6812-Strips connected to Wemos D1 mini. Everything was working fine with my older Version of diyHue and the "SK6812HueStrip.ino" (the one from the mariusmotea-repo). However, with the new Generic_SK6812_Strip.ino the wemos D1 mini is not broadcasting with the SSID set in the light_name variable. Instead it broadcasts with the SSID "ESP-C760DC". When connected to this network, nothing happens. I also tried with the Generic_WS2812_Strip.ino, but the result is the same.
When trying again with the old "SK6812HueStrip.ino" the strip is working with the Light GUI, but connection to the diyhue-bridge is not possible. I think it has to do with some name-variable.
Is there anything i'm missing out?

kind regards,
roman

romanlucas, 2019-03-23:
Hey again,
i'm going to close this issue. Found the mistake myself.
By using serial monitor in the arduino ide i could find out that it was failing to mount the file system. Turns out in Arduino IDE --> Tools --> Flash Size --> 4M (no SPIFFS) was selected. Changing this solved the problem. The Wemos D1 Mini broadcasted with the right SSID and the SK6812-Strip is working flawlessly again!
Maybe the hint with the SPIFFS (still no clue what this means, to be honest) could/should be mentioned in the Lights-Docs. As a complete noob to all this stuff i really wasnt aware of anything like this could solve my problem.

kind regards,
roman

closed 2019-03-23 by romanlucas

black-puppydog, 2019-05-29:
Just had the same happening to me. Couldn't reach the D1 Mini via http, but could ping it. I set the flash size to 4M (1M SPIFFS) and that seems to work now. Thanks @romanlucas for writing it down!