diyhue/Lights #32: Change Color Order

ryancasler, 2019-03-23:
Is it possible to change the color order in any of the sketches?  I have tried the SK6812 and WS2812 Sketches and I can't seem to find an easy way to change the color order.  I am finding the my Red and Green are exactly opposite each other.  So, setting the light color to green gives me total red and red gives me total green.  Blue is correct.  I have tried reversing the "RGBColor red" and "RGBColor green" and that doesn't seem to change anything.  Any assistance would be appreciated.

Mevel, 2019-03-23:
search for `NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod>* strip = NULL;`

Change to `NeoRgbwFeature`

Reference : [Wiki of Neopixel Library](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neo-features)

Thats how you change the order of the LED Colors.

ryancasler, 2019-03-23:
Thank you!!!  That worked perfect.  It was easier to do a find/replace because the sketch does have 2 references to NeoRgbwFeature or NeoRgbFeature depending on the sketch.  (just in case anyone finds this in the future).

closed 2019-03-23 by ryancasler