diyhue/Lights #33: ESP8266(Nodemcu 1.0) Generic_RGB_Light Pin D3 bug

m50008, 2019-03-27:
Hello,
i think there is a problem with the pin D3 (GPIO 0).
When setting pin d3 as output for the strip 
`pins[] = {D1,D2,D3};` in line 37 the output of the pin stays high, even when the strip is turned of using the hue app.
The other pins are working fine but pin d3 stays at 3.3v.

Mevel, 2019-03-27:
GPIO 0 has a "special" function. it is needed to put the ESP8266 into Flash mode. It has a internal Pull-up Resistor. 
GPIO 0 = LOW --> Flash mode.
I suggest using a different GPIO as per default in the Sketch!

m50008, 2019-03-27:
Is there a chanche to user GPIO 0 as output pin?
I have designed a circuit board(forgot the use of GPIO 0) and the output is now connected to this pin

mariusmotea, 2019-03-27:
i believe PWM is disabled when it is set to value 0. In your case when you set `digitalWrite(pin, LOW)` the pullup resistor is somehow canceled.

m50008, 2019-03-30:
Does `pins[] = {D1,D2,D3};' in line 37 change the pins? I think even when writing different pins there pin D0-D2 is setup as output

Mevel, 2019-03-30:
Pins are assigned without D, just gpio pin number. Ex ` pins[] ={5,12,13} `

m50008, 2019-03-31:
But even so, the output stays  the same. I can write every pin there

m50008, 2019-03-31:
Now everything works, when deleting lines 421-423 and lines 658-660. 
Now i can define the pins as i want, the output of my print method gives the pins i have declared and the led strip works, even with pin D3 or gpio0

mariusmotea, 2019-04-03:
i believe the issue is that you flash the firmware without to choose SPIFFS. If you want to give it one more try select on flash type 4M with 3M SPIFFS

closed 2019-04-14 by stale[bot]