diyhue/Lights #35: ESP8266 NodeMCU WS2812 not working

OtavioMiguel19, 2019-03-31:
I was trying to connect the led strip WS2812 to the ESP8266 and it didn't work. I used an other library (Adafruit_NeoPixel) to set manually a color and I did set static colors to it. It worked, but the original code wasn't working. Does it need any other config to work or is it really a bug?

Mevel, 2019-03-31:
No additional changes (in the code) are required.
Make sure you follow these steps after flashing the firmware.

- Configure Microcontroller by connecting to its Wifi, adding Credentials for Wifi Network. (ESP populates its own WiFi AccessPoint)
- brows to `192.168.4.1`
- Add new created diyHue Lamp within APP by searching for new Devices.

mariusmotea, 2019-03-31:
It is mandatory to use exactly the same libraries and eventually same version of them.

OtavioMiguel19, 2019-03-31:
It that you said was working fine. The ESP8266 have connected to the bridge successfully.
The problem is with led in this "lamp". It wasn't working. In `192.168.4.1` we clicked in ON and OFF and nothing happened.

The led strip continued always off, no matter what. I tested it in all the possible outputs, and the strip was still off.

That's why I told you about the other library I used. With that library, I could set a color in the led strip to test the output. With the default library, none of the outputs controlled any led of the led strip.

> No additional changes (in the code) are required.
> Make sure you follow these steps after flashing the firmware.
>
> * Configure Microcontroller by connecting to its Wifi, adding Credentials for Wifi [...]

romanlucas, 2019-03-31:
hi!
hope its ok to join.
im also experiencing issues with ws2812.
when connecting a wemos d1 mini (flashed with "Generic_WS2812_strip.ino")  to the ws2812b-strip, im able to configure the microcontroller and also add it to the hue app. but its not possible to correctly apply any changes in color/brightness or switching on and off. sometimes nothing happens, sometimes wrong number of leds with wrong colors are shown etc.
flashing the same wemos d1 mini with the "generic_SK6812_strip.ino" (and, of course, connecting it to a sk6812 strip) is working fine!
trying another ws2812b-strip (known as working with an arduino nano) gives the same results, so i dont think its a defect of the strip.
maybe im missing something out!?!

Pantastisch, 2019-04-06:
I was observing flickering and random colors, when the light and pixels counts do not fit or the cables to the strips are too long. Can you check that please?

romanlucas, 2019-04-10:
hey,
due to @Pantastisch's mentioning of too long cables, I redid all the wiring. Turns out the problem was a diode. Im fairly new to electronics... however placing the diode between RX of the wemos and a Resistor connected to Din of the Strip would case flickering and random colors. Placing the Diode after the resistor and connected to Din works just fine.
so  for my Part of this Issue: Problem solved. Lots of Thanks!

closed 2019-04-19 by stale[bot]

Ro4cHii, 2020-07-18:
> I was observing flickering and random colors, when the light and pixels counts do not fit or the cables to the strips are too long. Can you check that please?

I am experiencing a similar problem.

I use a WS2812B along with my NodeMCU and for the first couple of minutes, all 155 LEDs seem to be working just fine. I can apply scenes, switch colors, change brightness, you name it.

After some switching off/on of the individual lights, they start to flicker, then most of the pixels switch off and some random pixels switch on and stay this way even when I switch off the light via Hue app or diyhue API.

There seems to be nothing I can do to "repair" this except switch off my PSU and leave it this way for a couple of hours.

Coming from the RX pin, I am using about 80cm of 24AWG to go to the LED strip's DIN, VCC and GND. Is this too thin?
I already tried connecting a 220 Ohm resistor between the RX pin and the DIN since I read that multiple times online but that does not help either.

Does anybody know what else I can try?

ryancasler, 2020-07-23:
80 cm of wire is a long run even if you were using larger wire.  Is there no way that you can have the NODE MCU closer to the strip?  I make it a rule to have the controller no more than 1 foot from the board.  

Are you sure that your LEDs are 5v and not 12v?  Are you using a level shifter to change from 3.3 to 5v?  How many LEDs per meter is your strip?  155 LEDs is a long run.  You most likely need to add an additional 5v line in at the half-way point of the strip to provide enough power for the whole strip to run.  What amperage is your PSU?  Is it supplying enough amperage?

Why are you connecting a resistor between RX and DIN?  There should be no resistor there.  That will cause a drop in the current and cause the signal to be degraded tremendously.  You definitely do not want that.  You only need to use a resistor if you are running a regular LED, not WS2812 leds.  They have all the necessary resistors built into the chip.