diyhue/Lights #39: Travis compiled firmware not working

mariusmotea, 2019-04-10:
When i update the light using firmware from the repo the light is connecting to wifi but the light webserver is not responding, and the only way to recover the light is to flash it using USB port.

cheesemarathon, 2019-04-11:
Could this be an issue with SPIFS. When you upload over USB you set the SPIFS value in the IDE. But is this done when you an OTA update?

cheesemarathon, 2019-04-11:
@mariusmotea Try this file: https://github.com/diyhue/Lights/blob/cheesemarathon-patch-1/Arduino/bin/test/Generic_RGBW_Light.ino.bin
It's built a bit differently.

If you need me to build you a different sketch for you to test with, just shout.

mariusmotea, 2019-04-21:
@cheesemarathon i'm unable to test that firmware, my RGBW bulb has currently 512Kb flash. Can you generate the same for WS2812 strips?

I saw all firmware in main repo were generated again 11 days ago, i test the WS2812 version but still no luck.

cheesemarathon, 2019-04-21:
Just rebuilt it, in the cheesemarathon-patch-1 branch. Here: https://github.com/diyhue/Lights/tree/cheesemarathon-patch-1/Arduino/bin/test

mariusmotea, 2019-04-21:
I test now, is still not booting after uploading this firmware.

cheesemarathon, 2019-04-21:
That's very odd. I don't see how it's building any different. But I'll do
some testing on a local machine and see if I can spot any differences.

juanesf, 2019-04-25:
According to a collaborator of tasmota firmware says the following:

A short tip about:
- esp core 2.4.2
I'm from tasmota (github) and we know also things in the 2.4.2 core. Not always visible and don't come up on every used esp8266 hardware>
- The 2.3.x is working very well and we use it for the default binaries. All other versions like 2.5.x and alphas 2.6.x are for test use only.
So far my 2 cents..

mduerager, 2019-07-15:
The last adruino sketch file does not work when flashed on my board with ESP8266 on it. The older version on the old diyhue-repo does work like a charm. Can´t connect to 192.169.4.1 as the server refuses connection. Please have a look because this shouldn´t happen in an updated version...

mariusmotea, 2019-07-15:
Did you flash the board with SPIFFS  storage?

mduerager, 2019-07-16:
Yep. There is also something wrong with passing the LightName, it does compile, but there is some warning whilest it. Have to look when i am home again.

mduerager, 2019-07-16:
Compiling the Sketch for RGB-CCT Lights (latest sketch version) outputs:

\Generic_RGB-CCT-Light.ino:37:19: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 char *lightName = "New Hue RGB-CCT light";

mduerager, 2019-07-16:
Ok guys, i just reflashed the lateset firmware with nodemcu PyFlasher and now it works. Totally strange, i did nothing other yesterday, and the web-connection didn´t work...

gsg2016, 2019-07-17:
Hi Guys, This project is Awesome when it works. Has anyone else had this issue.  sk6812_wwa.ino works and connects to the network, but non of the newer sketches. eg. Generic_SK6812_Strip.ino  thanks for your hard work.
[image]

mariusmotea, 2019-07-17:
In new version is mandatory to have SPIFFS filesystem. Choose 2M SPIFFS in Arduino on board menu.

gsg2016, 2019-07-17:
oh! Thank you

cheesemarathon, 2019-10-03:
I'm going to close this issue for now. Travis builds have not worked for some time now, and I'm sure command-line building for Arduino sketches has changed a lot since I wrote the Travis scripts. I'll write something new when I get time.

closed 2019-10-03 by cheesemarathon