diyhue/Lights #43: Docu about entertainment Upd packet

obelix05, 2019-05-03:
Hello. Is there any documentation about the upd packet for the entertainment feature? Or can someone briefly explain the packet structure to me? I want to add the feature that only the selected entertainment lights will switch to "entertainment-mode". But I don't know reallly how the packet looks like.
Greets,
obelix05

mariusmotea, 2019-05-03:
Hi,

Entertainment mode works this way:

Entertainment app broadcast udp frames acording to entertainment protocol. This use a dtls encryption that is not available in python libraries, for this reason we use a proxy udp server compiled in c++ that is started only when entertainment mode is enabled and only purpose is to remove encryption and send raw udp data to hue emulator (python based) on secondary port 2001/udp listener (not remember exactly if this is the port). At this point the hue emulator parse the data to see what type is the light that need to receive the request. If this is an esp8266 based light with dedicated udp server for entertainment mode the rgb data of the frame is forwarded there, otherwise conversion is applied and delay added because all other types of lights are not able to process light data so fast. ESP8266 light receive all 24 updates per second, all other types of lighs receive just one update per second. I believe Yeelight receive more updates (~2) per second, i need check as i don't remeber exactly.

obelix05, 2019-05-03:
Hi.
Thanks for quick reply.
I'm interested in this three lines of code from "Generic_WS2812_Strip.ino"
```
    for (uint8_t i = 0; i < packetSize / 4; i++) {
      lights[packetBuffer[i * 4]].currentColors[0] = packetBuffer[i * 4 + 1];
      lights[packetBuffer[i * 4]].currentColors[1] = packetBuffer[i * 4 + 2];
      lights[packetBuffer[i * 4]].currentColors[2] = packetBuffer[i * 4 + 3];
    }
```
What does the packetBuffer look like if there is a unregistered light?
Imagine my diyLight has 3 Lights and i want to have entertainment on light 1 and 3, light 2 should stay in the "normal" mode. Is the packetBuffer for light 2 None or 0 or sth. like this?
The question comes up because I don't have a really good debugging idea for the lights on an ESP8266.

mariusmotea, 2019-05-03:
But your issue must not be an issue. If you exclude lights from entertainment area will simply skip that lights. I believe you use ws2812  strip.

obelix05, 2019-05-03:
Hi. Yes you're right. It is not an issue, but rather a general question. I didn't know where to ask my question.
And I'm sorry for my question. I didn't see any lights left out that weren't configured. And I had a bug in my code. Thanks for your advice. Sorry to disturb you.
Anyway: Thank you so much for your great work.

closed 2019-05-03 by obelix05