diyhue/Lights #45: New feature: Automatically merge left over Neopixels into new group

jnslsrwy, 2019-05-16:
Thank you for this amazing project! I successfully created a DIY Neopixel light with the Wemos D1 mini and a WS812 strip. But I ended up at a total number of 142 pixels. After setting everthing up 140 of the pixels lit while the last two remained off. I figured out that the script divides the total number of pixels by the number of virtual lights. Unfortunatley is 142 only dividable by 1, 2, 71 and 142. Therefore there will always be pixel in that strip which wont light correctly.

Would it be possible to let the script merge any pixels that are left over after creating the virtuals lights from the whole strip into a new virtual light?

mariusmotea, 2019-05-17:
Hi,

I believe the simple solution is to increase the number of leds until you divide the number of virtual lights you want. There will be no problem if the extra leds are missing.

jnslsrwy, 2019-05-18:
Wow, I could have thought of that myself :p. Thank you for your support!

closed 2019-05-18 by jnslsrwy