diyhue/Lights #46: --Question of compatibility--

OtavioMiguel19, 2019-05-19:
Is **Utorch BW - 5 E27** compatible with this system? If not, could be a good improvement of the system add a compatibility, 'Cause it is a very cheap bulb ($9).
Unfortunately, it has not a ESP board inside. That's the problem and that's why I'm asking it here..

https://www.gearbest.com/smart-bulbs/pp_009427364160.html

mariusmotea, 2019-05-19:
But what controller it has? ESP8266 is one of the most cheapest wifi controllers on the market and it must be present in most cheap bulbs.

cheesemarathon, 2019-05-22:
If it has an ESP8266 in it (couldn't find any confirmation of this online) then yes we have firmware that can be flashed to it (with slight modification). If it doesn't use an ESP8266 then it's unlikely that we can support it. I couldn't find any documentation of its control methods and there doesn't seem to be an API. So without buying one and spending considerable time reverse engineering it, it won't be supported.

closed 2019-05-22 by cheesemarathon