diyhue/Lights #47: Offline CSS

wolfi1k, 2019-05-19:
I am using the Generic_WS2812_Strip with a separate Wi-Fi from the raspberry pi without any internet access. Since the CSS file are an external resource the web interface can not load. I do not want to integrate the lights into my main wifi and also do not want to give the lights access to the internet. Is there any option to store the CSS files on the device itself?

Mevel, 2019-05-19:
One solution would be to make The css available within your network, adapt the code for esp and flash the "offline only" firmware.

mariusmotea, 2019-05-19:
is your wifi router able to override DNS entry? Because you can point it to a local ip (can be also the raspberry as it has internal webserver).

wolfi1k, 2019-05-19:
Hosting it on the raspberry might be a work around. Thank you very much

mariusmotea, 2019-05-19:
Can be hosted also on the light itself, check my power monitor project where css and js files are served from SPIFFS of the ESP.

cheesemarathon, 2019-05-22:
This may be an improvement we should make to the project. I'm not sure which option is the best but we could either:

1. Host it on the same device as diyHue is installed. Package it with the diyHue install.
2. Serve the files from the SPIFFS on the ESP

Only issue with number 2 I can see is running out of room on SPIFFS
What do we think?

mariusmotea, 2019-05-24:
Yes, ESP8285 has only 1MB flash and update require at last 350Kb free space. Moreover the CSS and JS files loading speed from ESP is very slow compared with our CDN.

mcer12, 2019-06-02:
What about this. There can be an optional setting in WIFI manager to add bridge IP address. If added, it would download the files from the bridge instead of external CDN.

closed 2019-06-11 by stale[bot]