diyhue/Lights #52: Hue Entertainment not working

mduerager, 2019-07-25:
I don´t know exactly why hue entertainment doesnt work (i can setup an entertainment area, but lights don´t recognize color changes), but i get following error when running in debug mode:
Is it possible this causes the problem?

Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/hue-emulator/functions/ssdp.py", line 28, in ssdpSearch
    data = data.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xdb in position 377: invali                         d continuation byte

mariusmotea, 2019-07-26:
That error is not related to entertainment but to ssdp discover process that probably received an unexpected search string. What types of lights are you using in entertainment and what PC application?

mduerager, 2019-07-26:
For now, i use an esp8266 controlled rgb-cct led stripe. As far as i remember, entertainment did work a while ago, but then i reinstalled the emulator and since then, the light stays green and doesnt flash when tested in the official hue app setup. I tried hue sync on windows pc, it does connect to the bridge, but no function on the stripe.

Fisico, 2019-07-26:
i have the same problem since i upgraded to light version 3.0 for the ws2812b strips. This version somehow is not working anymore with hue entertainment.

Also i had to set the red, green and blue multiplier to 100 to get full brightness.

JohnBFrm, 2019-07-27:
I can confirm that it isn't working here either.
ESP8266-based
Using a strip based on WS2812b

Changing colors via the Hue app's color wheel works great, but testing the lights in the entertainment setup shows no change in color. They simply stay green the whole time.
Hue Sync doesn't work at all due to the above.

mariusmotea, 2019-07-27:
If you downgrade the light to previews version it is working? You upgraded to new version or is the first flash of this light?

Fisico, 2019-07-27:
@mariusmotea it is also the case if i erase the flash and reflash 2.0 versions. Tried some older version always the same. The strange thing is, before i reflashed the wemos' to 3.0 yesterday all was working fine. Afterwards it's not working any more even after flashing back to 2.0 again.

Nor the diyhue bridge nor the app got an update yesterday so this is very strange.

mariusmotea, 2019-07-27:
I will try to replicate in a few hours.

JohnBFrm, 2019-07-27:
> I will try to replicate in a few hours.

I can try downgrading.
I've modified the code to print out any exception, and it's timing out on the service the entire time ("timed out"). When attempting to run the setup for Entertainment via the Hue app, it instead gives
`[Errno -2] Name or service not known
`

Fisico, 2019-07-27:
@JohnBFrm 
maybe make a pullrequest with your fix? or is this just a quick "hack"

JohnBFrm, 2019-07-27:
> @JohnBFrm
> maybe make a pullrequest with your fix? or is this just a quick "hack"

It might be good to review what has changed first, as the problem could be potentially fixed higher up. The above function is passed the address of the lights from HueEmulator3.py, when it creates a new threads and calls entertainmentService, passing it lights_address from config.json, which in turns has the address of the lights including the port (":80").

If this has changed recently, it might've been overlooked.

mariusmotea, 2019-07-27:
If i'm not wrong the timeout is usefull there because once the entertainment app stop the broadcast i need to set the entertainment service as ready for new connections. There is a key in hue api that reflect the service status and can be true or false.

JohnBFrm, 2019-07-27:
> If i'm not wrong the timeout is usefull there because once the entertainment app stop the broadcast i need to set the entertainment service as ready for new connections. There is a key in hue api that reflect the service status and can be true or false.

Without being too read up on the Hue API, I understand the reasoning, however, entertainmentService() never really stops running, and is constantly listening for traffic, and is therefor constantly throwing out the quiet exception.
As far as I can see, we hit a timeout at:
```data = serverSocket.recvfrom(106)[0]```
which triggers an exception, dropping us down to:
```if syncing:```
which we've already got set as false, then the function is run all over again.

Do you perhaps mean:
``                for group in groups.keys():
                    if "type" in groups[group] and groups[group]["type"] == "Entertainment":
                        groups[group]["stream"].update({"active": False, "owner": None})```
We're not able to reach this unless we've begun running the rest of the function and throw an exception after timing out OR already received data  (as was happening before). As far as I can see in the code, we should only really hit a timeout exception during `data = serverSocket...` which shouldn't be necessary, but I might be wrong!

Just some thoughts!

Fisico, 2019-07-29:
@JohnBFrm @mariusmotea any news for this issue?
Would love to use hue entertainmain again :D

mariusmotea, 2019-07-30:
I debug this issue and is not good. The problem seams to be in the C compiled entertainment service that was not made by me. I will try to recompile it, maybe it will start working.

```
  . Performing the SSL/TLS handshake... hello verification requested
  . Waiting for a remote connection ... ok
  . Performing the SSL/TLS handshake... ok
    [ Protocol is DTLSv1.2 ]
    [ Ciphersuite is TLS-PSK-WITH-AES-128-GCM-SHA256 ]
    [ Record expansion is 37 ]
    [ Maximum fragment length is 16384 ]
  . Verifying peer X.509 certificate... ok
2019-07-30 10:31:20,318 - root - INFO - Entertainment Service was syncing and has timed out, stopping server and clearing state
 mbedtls_ssl_read returned -0x45
 interrupted by SIGTERM (not in net_accept())
  . Cleaning up... done.
```

mariusmotea, 2019-07-30:
i made a mistake, seams the issue is that now the port is added in lights_address. I will fix it.

mariusmotea, 2019-07-30:
Fixed with commit https://github.com/diyhue/diyHue/commit/4af4993c28c83fa5d6ca3e5caba8e16c0cd5351c

For all other users a new light search may be needed in order to add the port in light address.

closed 2019-07-30 by mariusmotea