diyhue/Lights #53: Multiplier has to be atleast 1 to get a light.

Fisico, 2019-07-26:
After flashing my wemos to light version 3.0 with the ws2812b sketch, I had to set the red, green and blue multiplier to atleast 1 to get any light out of the strip and to 100 to get full brightness. Shouldn’t this be a default value to be set in the code?

mariusmotea, 2019-07-26:
Default are 100, but in case of upgrade it is not working. I will look for a fix to this bug.

mcer12, 2019-07-26:
When you make an OTA update or update without complete memory wipe, your old saved config file stays intact. There is no check if json contains the values, that means if new values are added in an update like the multiplier, it is undefined.

Fisico, 2019-07-26:
I tested it and after a wiping the flash and reflashing the v3 firmware the multiplier value is 100 so this is not a problem than.
Should I close the issue?

Fisico, 2019-07-29:
fixed with https://github.com/diyhue/Lights/commit/0462e7f7c3529f681b70b65a6d5fc1c4b7227cda

closed 2019-07-29 by Fisico