diyhue/Lights #54: Documentation says Update -> Arduino OTA | But that is missing in each sketch ;-)

JorinL, 2019-09-06:
Hey there!

Great work at all! 
I just found the docs saying "use Arduino OTA" but this is not used inside most sketches (I did not check all of them).

Is that correct or was there a problem with OTA? Otherwise I'll engange it for the generic rgb one.

mariusmotea, 2019-09-06:
Whre OTA is missing was introduced the update method via /update path of the light. This is because bridge can check the version and update automatically the lights.

JorinL, 2019-09-06:
Ah so a precompiled binary is uploaded via /update on the lights.
How does that update with the bridge work? :-)
Is there some doc I dont know? (searching while typing).

cheesemarathon, 2019-09-06:
Update via the bridge currently does not work. All that will do is change the version number the lights report so that the hue app thinks they are genuine, upto date bulbs.

JorinL, 2019-09-06:
Ah ok - Thx for explanation.
Well OTA could be corrected at the dosc ;-)

I close this here.

closed 2019-09-06 by JorinL

mariusmotea, 2019-09-06:
You can export precompiled firmware from arduino directly and upload it via /update path.

JorinL, 2019-09-06:
> 
> 
> You can export precompiled firmware from arduino directly and upload it via /update path.

Yep.