diyhue/Lights #57: RGBW Hue Ring?

robotica72, 2019-09-11:
Any chance someone could add the FW that changes the original Hue Ring code from RGB to RGBW?

mariusmotea, 2019-09-17:
This is very easy to do, just add the 4th channel in the sketch, but be aware that ring sketch is not updated and miss some new futures.

robotica72, 2019-09-24:
Thanks - I thought it was more complex than just adding the 4th channel.   Are there any planne updates to the Ring code for the newer features?    I havent looked at any other code yet - so I may be able to use something else for my projects.

mariusmotea, 2019-09-25:
Is complicated to upgrade this sketch because in new version all settings need to be migrated in the web gui and stored in SPIFFS.

closed 2019-10-03 by cheesemarathon