diyhue/Lights #58: Lights won´t be discovered

Jibbonator, 2019-09-20:
Hey Guys, I use a Wemos D1 mini With WS2812B Led Stripe.
I have flashed the lights gui. (Lights/Arduino/Generic_WS2812_Strip/)
I can change the color over the web interface, but when i try to discover it in HUE app,
nothing ist found. I have several Tasmota devices that are found.

Do i have to config something to make it work?

Can someone help me.

philipphunger, 2019-09-20:
I have the same problem, in the log it says <ip> unknown device

mariusmotea, 2019-09-22:
Can you place the light near the wifi router? If the latency is too big the light will not be detected. Also you can see what is going on and put also the the debug output of the emulator during light scan.

mariusmotea, 2019-09-22:
Forget one thing, in webgui only the lights added in groups are displayed, so new detected lights will not be there.

Jibbonator, 2019-09-22:
My bridge is connected with lan an the ping from bridge to wemos is stable from 1,5 - 8,5ms.

Here is my log:
all i can sse is that it says unknown device.
[log.txt](https://github.com/diyhue/Lights/files/3639911/log.txt)

mariusmotea, 2019-09-23:
I don't understand why, but the light ip is detected when probe for yeelight and epshome are made (for this reason you see the ip 192.168.0.60 in the lists) but when the probe for native lights is made the ip is not displayed:

Search for  `"GET /detect HTTP/1.1"`

I don't see `.60` in the list:

```
2019-09-21 12:23:36,421 - root - INFO - [
    "192.168.0.1:80",
    "192.168.0.2:80",
    "192.168.0.3:80",
    "192.168.0.5:80",
    "192.168.0.12:80",
    "192.168.0.13:80",
    "192.168.0.25:80",
    "192.168.0.28:80",
    "192.168.0.39:80",
    "192.168.0.57:80",
    "192.168.0.58:80"
]```

Jibbonator, 2019-09-23:
i think i might found a solution. i´ve disconnected my device 192.168.0.39 and now it works. its a NAS. weird

philipphunger, 2019-09-24:
If it is the nas, try searching the lights with the IP-range Argument on the cli. That works for me as i have a esxi server on the network.

closed 2019-10-03 by stale[bot]