diyhue/Lights #62: Glitches when using over 200 leds

xcruell, 2019-11-07:
Hi,
im using ~220 WS2811 LEDs in my living room, when i set a value of over 200 LEDs in the web interface, everything just glitches out.
I cant turn the light off anymore, color changes have big delays and often end in a wrong color and similar stuff like this.
But when i set the value to exactly 200, everything works perfectly fine.. (except one conor of my ceiling is dark ^^)

Im using the "Generic_WS2812_Strip.ino" (i changed the GRB to BRG) and the Hue Essentials App.

regards

mariusmotea, 2019-11-07:
I controll 300 leds with no issue. Most likely you have a power supply with less than 5A and the leds receive insufficient power.

xcruell, 2019-11-08:
Yeah, im using a 4A PSU.
But even if i put the LEDs on red and like 50% brightness, this happens.
Im going to replace it with a 16.5A PSU from my old Xbox^^
I saw that the middle of the strip suffers from a voltage drop, since they only shine red, when i put them all on white. I will tap a extra cable there.

mariusmotea, 2019-11-08:
In your case must be a hardware issue as the solution is tested with more than 300 leds and with proper power supply it is working fine. Power the strip directly, not via the esp8266 board.

xcruell, 2019-11-08:
The WS2811 LEDs are 12v, i cant Power them through the ESP.. 😂
Im using a wemos D1 Mini, a sn74hct245n Level shifter and a step down to Power the ESP with 5v.
Im going to change the psu and add the Power lines in the middle tomorrow

Paalap, 2019-11-08:
Best approach is to inject the power on multiple points

Zappelphilipp, 2019-11-08:
i am using a strip with 600 LEDs without any problems :O

xcruell, 2019-11-13:
Im using a 16.5A 12V PSU now, im injecting power at 4 points and it still "crashes/glitches" when i put more than 207 Pixel.
When i change the Pixel Count to 208+, the last 13 Pixel (Pixel no. 208-220) are emitting light, even if i turn the strip off in the hue app.
i didnt change anything in the code, except the neopixelbus line (GRB to BRG)

When i change the value back to 207, everything works like a charm.

It really is the diyhue code, because i can use well over 300 with the neopixel library, without any problems what so ever.
I used different ESP8266s too

mariusmotea, 2019-11-13:
@xcruell it may be an issue with NeoPixelBus library for WS2811 . If there was an issue with WS2812 sketch believe me we will know it alreasy as there are hundreds of users that use this sketch with no issues. Please read the documentation of the library from [here](https://github.com/Makuna/NeoPixelBus/wiki) is possible to require more changes to the code and also because is a 12V led maybe you need to use a level shifter.

xcruell, 2019-11-13:
Hi @mariusmotea .
As i stated before, im using a levelshifter and i dont have any Problems with the neopixelbus itself. Only with the diyhue code.

mariusmotea, 2019-11-13:
did you tried to run the esp at 160Mhz?

closed 2019-11-22 by stale[bot]