diyhue/Lights #63: no webui

snickers2k, 2019-11-08:
Hey. Just uploaded your .bin file to my wemos. wifi shows up, i can connect - but there is no webui.
could you please update the .bin?  i have no clue of compiling the .ino with SPIFFS .. what ever this is

thank you

mariusmotea, 2019-11-08:
SPIFFS is an option you must choose in Arduino tools menu before you flash the firmware.

snickers2k, 2019-11-08:
oh, thank you for your super fast answer :D 
problem is, i have no clue of Arduino IDE .. i simply use nodemcu-flasher (or esptool) :/
there are no options. it's simply uploading the .bin to the esp , nothing more... thought thats why there is a pre-compiled .bin in your project ?

mariusmotea, 2019-11-08:
with esptool i think is complicated to set spiffs. Maybe the most easy way is to use Arduino IDE (is cross platform). On a basic search i found this [tutorial](https://github.com/toblum/McLighting/wiki/Software-installation). On one if the images you will see the option `Flash Size: "4M (1M SPIFFS)"`

snickers2k, 2019-11-10:
again, thanks for you
sorry, but i think, just like me, i guess there are many people out there having no clue about this ..
could there be another way? is it possible to use ESPEasy with diyhue (like via http requests)? or is it possible to add the spiffs part into the .bin directly? 

thank you

mariusmotea, 2019-11-11:
I don't think the spiffs is part of .bin file, and this is because when i try to upgrade the old firmware witch don't use spiffs with the new one via web update it did not work. Seams the SPIFFS partition must be created during initial flash.

snickers2k, 2019-11-12:
okay.. thank you again.

and there is no other way to keep esp's running with diyhue ?

mariusmotea, 2019-11-13:
espeasy support can be added by community if somebody with skills witch need this future will implement it, but i doubt this will happen as is more easy to use the firmware we provide for esp8266/esp8285 and have better support with all hue futures ready.

alexyao2015, 2019-11-18:
Esphome is supported and does not require spiffs. It's the best option imo

closed 2019-11-27 by stale[bot]