diyhue/Lights #65: BUG - Entertainment causes flashing white

stvnmbr1, 2019-11-19:
Wemos D1 Mini
WS2812 generic led strip

when using entertainment, at a random time the leds start pulsing white.
 controls keep working, turning the leds or room of turns off the leds but the pulsing stays.

unplugging and plugging back in seems to be to only way to resolve this

mariusmotea, 2019-11-20:
Are you sure the power supply deliver enough power? Do the issue with flickering ocur only when all leds are on at maximum brighness?

stvnmbr1, 2019-11-21:
Yes because my ring, who has for sure enough power also starts flashing at
the same time. Hue app shows them going on and off.

Seems to be random, caused by the hue sync app(or cpu load), will try to
figure out when it happens exactly.
Been running entertainment for 20mins at full brightness now and hasn't
happened again so far

stvnmbr1, 2019-11-21:
Redid the wiring, the flashing still appears but i think it just is trying
to tell that a light is not connected anymore.

stvnmbr1, 2019-11-21:
As in connection timeout

mariusmotea, 2019-11-21:
how many leds are you using and what is the number of virtual lights?

stvnmbr1, 2019-11-22:
60 leds, 4 lights, 2 transition leds
Redid wiring, psu directly to strip and data in to wemos.

You can reproduce this by starting hue sync and unplug one of the lights of
the entertainment room.
In my case it's bad wifi connection in a corner that is causing this.

closed 2019-11-26 by stvnmbr1