diyhue/Lights #68: no member named 'createNestedObject'

freakadings, 2019-12-09:
I was trying to compile the Arduino SK6812-Stripes Sketch and this exception came up:

`class ArduinoJson::DynamicJsonDocument' has no member named 'createNestedObject'`

I looked around the net and found the following:
"References were removed in 6.1.0"
https://github.com/bblanchon/ArduinoJson/issues/772

So i tried to install two older versions of ArduinoJSON (by Benoit Blanchon) 
6.0.1 -> Same issue
6.0.0 -> Same issue
5.13.5 -> "DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed."

I'm a little curious which library (version) you used :D

mariusmotea, 2019-12-09:
I made a test now with with Generic_WS2812_Strip sketch and last esp8266 board and libraries the compile process is successful. I used Arduino 1.8.6

closed 2019-12-12 by mariusmotea

freakadings, 2019-12-13:
Sorry for my late response.
Have you tried it with the "Generic_SK6812_Strip.ino", too ?

I removed the complete arduino IDE from my system and reinstalled it from scratch (win10, Arduino IDE v1.8.10).

What i did:
on win10:
installed: arduino-1.8.10-windows.exe

within the ide: 
added "http://arduino.esp8266.com/stable/package_esp8266com_index.json" to File -> Preferences -> Additional Baords Manager URLs

installed the "esp8266 by ESP8266 Community" Boards

Installed the following libraries:
```
"ArduinoJson by Benoit Blanchon Version 5.13.4" (as linked by you)
"FastLED by Daniel Garcia Version 3.3.2"
"NeoPixelBus by Makuna by Michael C. Miller (makuna@live.com) Version 2.5.4"
"rc-switch by sui77 Version 2.6.2"
"WiFiManager by tzapu Version 0.15.0-beta"
```

Selected Board: "LOLIN(WEMOS) D1 R2 & mini"

Opened a new scetch, and copied the content of 
`https://raw.githubusercontent.com/diyhue/Lights/master/Arduino/Generic_SK6812_Strip/Generic_SK6812_Strip.ino`
in there.

Tried to Compile and the following errors came up:

```
Arduino: 1.8.10 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Users\freakadings\Documents\Arduino\sketch_dec13a\sketch_dec13a.ino: In function 'void saveState()':

sketch_dec13a:401:3: error: 'DynamicJsonDocument' was not declared in this scope

   DynamicJsonDocument json(1024);

   ^

sketch_dec13a:401:23: error: expected ';' before 'json'

   DynamicJsonDocument json(1024);

                       ^

sketch_dec13a:403:24: error: 'json' was not declared in this scope

     JsonObject light = json.createNestedObject((String)i);

                        ^

C:\Users\freakadings\Documents\Arduino\sketch_dec13a\sketch_dec13a.ino:403:24: note: suggested alternatives:

In file included from C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:0,

                 from C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266HTTPUpdateServer\src/ESP8266HTTPUpdateServer.h:4,

                 from C:\Users\freakadings\Documents\Arduino\sketch_dec13a\sketch_dec13a.ino:5:

C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/detail/mimetable.h:15:3: note:   'json'

   json,

   ^

C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/detail/mimetable.h:15:3: note:   'json'

[... 522 lines cut ...]
Multiple libraries were found for "NeoPixelBus.h"
 Used: C:\Users\freakadings\Documents\Arduino\libraries\NeoPixelBus_by_Makuna
Multiple libraries were found for "SPI.h"
 Used: C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\SPI
Multiple libraries were found for "WiFiManager.h"
 Used: C:\Users\freakadings\Documents\Arduino\libraries\WiFiManager
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\freakadings\Documents\Arduino\libraries\ArduinoJson
Multiple libraries were found for "ESP8266mDNS.h"
 Used: C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266mDNS
Multiple libraries were found for "ESP8266HTTPUpdateServer.h"
 Used: C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266HTTPUpdateServer
Multiple libraries were found for "ESP8266WebServer.h"
 Used: C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer
Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\freakadings\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\DNSServer
exit status 1
'DynamicJsonDocument' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

I don't know anything else i could try :/

mariusmotea, 2019-12-18:
Please upgrade ArduinoJson by Benoit Blanchon Version to latest version 6.xx . Can you remember where i recommend that version so i can fix the documentation?