diyhue/Lights #71: Cannot open Webinterface with Generic_RGB_Light

Gaming12846, 2019-12-29:
I set up the Generic_RGB_Light and connected to the wifi, but when I try to open the web interface I get a timeout and my Hue emulator cannot find the RGB stripe.

Mevel, 2019-12-30:
Could you try a different sketch and see if the problem persists? Do you use static ip or DHCP?

Gaming12846, 2019-12-30:
Other Stetches like Generic_WS2812_Strip working perfectly, i can open the webinterface and the hue app detects the Light. I use DHCP on the Sketch and set a static ip in my Fritz Box. Its work fine for  	Generic_WS2812_Strip.

Gaming12846, 2019-12-30:
I solved it, i switched to esp8266 version 2.4.1 with these settings:

    Set the CPU frequency to 80 MHz
    Set the Flash Size to "4M (1M SPIFFS)". <-- Double check this!
    Set the upload speed to 115200 (you could try to set it higher, but it could cause problems).
    Set "IwIP Variant" to "v2 Higher Bandwidth".

closed 2019-12-30 by Gaming12846