diyhue/Lights #74: Documentation Is Incorrect for Lights API

ryancasler, 2020-01-13:
I am having trouble trying to figure out the Lights API as documented here: 

https://diyhue.readthedocs.io/en/latest/lights/diylights.html#lights-api

It appears that data is out of date.  What protocol/path should I be using for the latest Lights Sketches?  It appears that /state is the correct path but all i ever get back is "fail", so I'm not sure what I'm doing wrong.

mariusmotea, 2020-01-13:
Please open developer tools in any browser to see the HTTP PUT request data. Now is used json format like hue api.

ryancasler, 2020-01-16:
I have tried that.   I sent a PUT message to this address:

http://192.168.1.125/state?light=1

With this JSON body:
{on: false,bri: 10, xy: [0, 0], ct: 370, hue: 5868, sat: 254, colormode: "ct"}

And nothing changed.  Nothing changes on the light no matter what I send.  So, what am I doing wrong?

closed 2020-01-16 by ryancasler