diyhue/Lights #75: Using RGB and W channel together in color mode?

Thauxeeb, 2020-01-30:
Hello!
First of all, this project is awesome, and thanks for making it!
Currently, I've been using the Generic_RGBW_Light, with a really great RGB&NatWhite strip, the white mode in the app works perfectly, however, if I switch to color mode, the White LEDs doesn't doing anything.
It would be nice to make for example some kiwi color by green channel AND white channel, rather than mixing in the white from the RGB LEDs. 
I think the original Hue Strip works this way, but it even uses the 2 White channels.

Thank you!

mariusmotea, 2020-01-30:
Hi,

I tried to made it like that, but the mathematics are complex to convert it from CIE colorspace values (used in general by hue apps) to RGBW. From HUE/SAT to RGBW is easy task but i prefer to not have different behaviour for the two color modes.

TravisLite, 2020-02-06:
This would be nice to have as a compilation option for those of us who do
prefer to have the option.

The whole reason I bought RGBW strips was to have better color mixing,
rather than relying on the RGB leds alone for color. Am I misunderstanding
something?

Thank you

closed 2020-02-15 by stale[bot]