diyhue/Lights #76: Would be nice to have a /clear to reset wifi settings

JorinL, 2020-02-12:
Hey there - I currently change my wifi setup and brought in that function:

```
  server.on("/reconfigure", []() {
    server.send(200, "text/html", "clear wifi config");
    delay(1000);
    WiFiManager wifiManager;
    wifiManager.resetSettings();
  });
```

To switch the wifi password.
Would be cool to bring it into the lights - like above or in another way :-)

mariusmotea, 2020-02-21:
Sorry for delay reply. But the WiFi Manager will not start the internal SSID for wifi reconfiguration if you change the password?

JorinL, 2020-02-21:
It looks like it does - but for me the solution doing so had much more comfort :-)

I could easily change the wifi password and then change it on the accesspoint and all connected.
Yes, it would also work by connecting the the phone to the strip or via notebook... but would have been more work :-D

closed 2020-03-01 by stale[bot]