diyhue/Lights #77: Use WS2812b on esp32

JoBobNenner, 2020-03-24:
Is it possible to run the "Light" sketches on an esp32 with *12b neopixels?
I've tried it, but the libs are different to esp8266.
Thanks for any replies.

mariusmotea, 2020-03-24:
I think you need minimal modifications for esp32 to work. First some libraries names are different (aka: webserver), then you need to read neopixelbus library modification for esp32 (if any).

JoBobNenner, 2020-03-24:
My second try was with the "ESP32_SK9822_FastLED_Strip"-Script
after several library installations ive fixed the ESP32 implementation.
I also modified lines 95 and 97, same code but without Clock-Pin, as is use 3 Pin ws2812b with GND DATA VCC. But it shows me the error:
`'HueApi' does not name a type`

mariusmotea, 2020-03-24:
Maybe https://github.com/linuzer can help you as he develope the code for esp32

closed 2020-04-02 by stale[bot]