diyhue/Lights #79: ESPHOME

gastonMM, 2020-05-25:
Hello,
I am trying to configure diyhue with ESPHOME, with the template you uploaded, but I cannot find the configuration, I was all Sunday trying to make it work, but I am not hitting the nail on the head.
If you have time, could you tell me what is wrong?

Thank you

`:
  name: tira_led_cubo_living
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret ssid_wifi
    password: !secret pass_wifi
  fast_connect: on
  reboot_timeout: 60s
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.1.212
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 192.168.1.1
  ap:
    ssid: "Tira LED Cubo Living Hotspot"
    password: !secret pass_ap

captive_portal:

api:
#  password: !secret pass_api
  
web_server:
  port: 80

logger:
  level: DEBUG

ota:
  safe_mode: True
  password: !secret pass_ota
  
sensor:
  - platform: wifi_signal
    name: "WiFi Tira LED Cubo Living"
    update_interval: 60min
    
light:
  - platform: rgbw
    name: "Tira LED Cubo Living"
    red: color_red
    green: color_green
    blue: color_blue
    white: color_white
    restore_mode: ALWAYS_OFF
    
output:
  - platform: esp8266_pwm
    id: color_red
    pin: GPIO12
  - platform: esp8266_pwm
    id: color_green
    pin: GPIO05
  - platform: esp8266_pwm
    id: color_blue
    pin: GPIO13
  - platform: esp8266_pwm
    id: color_white
    pin: GPIO15
    
text_sensor:
  - platform: template
    name: "light_id"
    id: light_id
    lambda: |-
      return {"esphome_diyhue_light;C4:4F:33:AE:68:79;cubo;255;255"};
    update_interval: 30s
`

mariusmotea, 2020-05-26:
Unfortunately i have no ESPHome lights. All i can tell you is that you must connect the light to your wifi network and perform light search from hue application. @alexyao2015 is the expert on this topic.

alexyao2015, 2020-05-26:
Find the docs here.
https://diyhue.readthedocs.io/en/latest/lights/esphome.html

alexyao2015, 2020-05-26:
Specifically, the issue here is the led names were configured improperly. Use the rgb and ct platform instead of rgbw.

closed 2020-05-26 by alexyao2015