diyhue/Lights #82: Wifi AP

gastonMM, 2020-06-12:
Hello, I have 3 ESP8266, with the Generic_WS2812_Strip.bin firmware, and after connecting to the wifi network, the AP continues with the captive page.

Also try Generic_WS2812_Strip.ino, with the same result.

gastonMM, 2020-06-12:
screenshot
https://drive.google.com/file/d/1htc1saRM0LcGGdBxrgkBWMXJJTjErGyS/view?usp=sharing

mariusmotea, 2020-06-16:
This happens if the wifi was not ready during ESP8266 boot. Is an annoying bug. 
The issue is with wifimanager and at this line:

https://github.com/diyhue/Lights/blob/master/Arduino/Generic_WS2812_Strip/Generic_WS2812_Strip.ino#L623

gastonMM, 2020-06-16:
Do you have any template without wifi manager? that allows to put ssid and password

try to replace it with the following data but fail to load it

`#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

const char* host = "192.168.1.1";
const uint16_t port = 3000;

ESP8266WiFiMulti WiFiMulti;

void setup() {
  Serial.begin(115200);

  // We start by connecting to a WiFi network
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid, password);

  Serial.println();
  Serial.println();
  Serial.print("Wait for WiFi... ");

  while (WiFiMulti.run() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  delay(500);
}`

mariusmotea, 2020-06-25:
Look here: https://github.com/tzapu/WiFiManager/

> //first parameter is name of access point, second is the password
wifiManager.autoConnect("AP-NAME", "AP-PASSWORD");

closed 2020-07-05 by stale[bot]