diyhue/Lights #83: ESPHome only compiles as RGBW

Cludch, 2020-06-15:
When trying to create an only RGB configuration, the compilation fails, because this line
https://github.com/diyhue/Lights/blob/master/ESPHome/AsyncUDP/diyhueasyncudp.h#L26 (in all three files) requires a white_led to be returned. However, this is not the case.

Therefore, a check should be add to check whether a white_led exists.
I haven't really worked with C++ but I might try this myself next week.

mariusmotea, 2020-06-16:
Did you test also with older wled library? Can you post the versions of libraries you are using?

Cludch, 2020-06-16:
I didn't use wled at all?

mariusmotea, 2020-06-16:
sorry, i was looking to your code and i misread  white_led as wled. Anyway i can test to compile on my laptop if you provide me your libraries versions.

Cludch, 2020-06-16:
I simply cloned the repo.
And then try to remove the `white_led` from the esphome yaml.
If you look at the linked code above, you will see that this cannot work and will throw an error.

Cludch, 2020-06-21:
I did some testing and realized there is no way to check whether the light exists.
Therefore, I guess, the best would be to just duplicate the header files. One for RGB and one for RGBW, which contains the `white_led`.

Cludch, 2020-06-28:
Not stale.

closed 2020-07-10 by stale[bot]