diyhue/Lights #86: ESP8266 NodeMCU: WS2812B suddenly stops working

Ro4cHii, 2020-07-18:
Hello there,

as mentioned in #35, I am experiencing a similar problem.

I use a WS2812B along with my NodeMCU and for the first couple of minutes, all 155 LEDs seem to be working just fine. I can apply scenes, switch colors, change brightness, you name it.

After some switching off/on of the individual lights, they start to flicker, then most of the pixels switch off and some random pixels switch on and stay this way even when I switch off the light via Hue app or diyhue API.

There seems to be nothing I can do to "repair" this except switch off my PSU and leave it this way for a couple of hours.

Coming from the RX pin, I am using about 80cm of 24AWG to go to the LED strip's DIN, VCC and GND. Is this too thin?
I already tried connecting a 220 Ohm resistor between the RX pin and the DIN since I read that multiple times online but that does not help either.

Does anybody know what else I can try?

romanlucas, 2020-07-18:
hi @Ro4cHii 

> Coming from the RX pin, I am using about 80cm of 24AWG to go to the LED strip's DIN, VCC and GND. Is this too thin?

just to be sure: does this mean you have connected the VCC and GND of your strip directly to 5V and GND of your NodeMCU?

Ro4cHii, 2020-07-18:
Hey @romanlucas,

correctly. Coming from my 6A PSU, I feed both the NodeMCU and the strip.

Ro4cHii, 2020-07-19:
I seem to have found the issue.
Luckily, I have another NodeMCU here that I could try out and with that one erverything is working as expected.

Most probably I made a mistake when soldering the level shifter onto the circuit board and there is a cold solder joint somewhere, so I'll just have to look :)

closed 2020-07-19 by Ro4cHii

ryancasler, 2020-07-23:
6A is not going to come even close to providing enough power for 155 WS2812B LEDs.  Each LED can consume up to 0.3 watts at 100% of R, B and G.  That means that 155 of them will use 46.5 watts.  That's 9.3 A of power.  Your PSU is most likely going to burn out or switch itself off.  You are pulling WAAAAAY too much power from it.

Ro4cHii, 2020-07-23:
It indeed is providing enough power, since I usually never go above 50% brightness and most of the time use a yellowish or blueish hue. 

Correct, both the microcontroller and the strip are directly connected to the PSU (with the strip being connected at both ends).
I re-checked the soldering on my board and indeed it was a cold solder joint between the level shifter and the data-out pin. After correcting it, everything is working just fine.

Anyway, what you are writing about the Wemos D1 sounds interesting. Can it output 5V signals or is it 3V3 as well?

ryancasler, 2020-07-23:
All ESP8266 are 3.3 v logic.  But you have to remember, for a digital signal, it is just 1's and 0's.  So, the signal can only be high or low.  Most 5v systems are built and are also 3.3v "tolerant".   Here's a SparkFun learning article that describes why you don't need the level shifter a log better than I can.

https://bit.ly/3jzAQ9p

Ro4cHii, 2020-07-24:
Well, I guess I just have to try, right?

The reason I initially used a level shifter is due to what is mentioned in this article: https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/

It states that a high is detected by the WS2812 at 0.7*Vcc which would be 3.5v and not 3.3v. However, it also mentions a simple hack without having to use a level shifter in case you run into any problems when feeding 3v3 data directly into the strip.

ryancasler, 2020-07-25:
Yeah, I don't know what they're smoking over at Hackday but I'd like some.  To be honest, I find a lot of their articles to be completely 100% wrong.  And this is definitely one of them.  The voltage you run the LEDs at will not affect the level at with the chip in the LED senses a 1 vs a 0 (high or low).  If you read through the comments you'll see why this isn't really a solution.

It all comes down to what you are going to be doing with these strips.  If they are meant to do high speed switching between colors where you have a LOT of leds on a single data line where you will frequently have two a different setting for every pixel in the run (like you would have for a big LED matrix display like this one.
https://youtu.be/zAU_l-wHP1Y
THEN you would need a level shifter because you are going to have to switch the LEDs so fast from one color to another to get it to look right.  But if you are just putting up some LED strips for biased lighting above your cabinets or along your floorboards, then you are not going to be doing any anything that would require you to have a level shifter.  Like i said, i have a bunch of LED strips around the house that I run directly from an ESP8266 and none of them have level shifters.  Because most of the time they are just static.  I only used the digital LEDs vs analog ones because I sometimes want to run animations or effects with them.  But nothing that is very taxing on the timing.