diyhue/Lights #89: Generic_WS2812_Strip WebInterface broken

cyptus, 2020-08-18:
The webinterface of `Generic_WS2812_Strip` script serves contents from `https` sites which does not work as the webinterface it self is only served by `http` (`mixed content`).
Especially contents of `https://raw.githubusercontent.com` are not allowed, as github serves them with response content type `plain/text` which is not accepeted by modern browsers for scripts or css.

Also i am not sure if the webinterface itself works like expected, as there are no values loaded and i did break the code twice by entering values and pressing save (boot loop after config load). I needed to clear flash storage afterwards.

Mevel, 2020-08-18:
This might be introduced due to #84 @dobby5 @alexyao2015

mariusmotea, 2020-08-18:
A public CDN set the mime type for hosted files, while github not. We need to swith to a public CDN

mariusmotea, 2020-08-18:
I just made the switch to diyhue.org CDN

closed 2020-08-18 by mariusmotea