diyhue/Lights #91: NodeMCU WS2812 won't show up

kompas1, 2020-08-29:
So I flashed the Generic_WS2812 with all customizations that needed to be done (the Pixel-count only) and I could flash it successfully. When I power the whole thing up i get a, some sort of "boot-animation" but if i try to connect to the hotspot of the node I can't because it didn't get shown even after 10 Minutes no sight of the hotspot. Has anyone had a similar problem?

mariusmotea, 2020-08-29:
Did you flash is with spiffs storage?

kompas1, 2020-08-30:
I've setup the Flash Size to 4MB(FS: 3MB) do you mean that? Or is there somewhere an other option i've not seen somehow

kompas1, 2020-09-04:
so after a lot of sleepless nights and a lot of fiddeling around i got it to work. Now I just have to connect it to the diyHue Bridge and everything is fine

closed 2020-09-04 by kompas1