diyhue/Lights #92: diyHue Bridge doesn't shows up

kompas1, 2020-08-30:
Sorry, It's me again. But this time with another problem:
So I tried both the manual and the automatic install, tried to connect to the bridge with the official App but the Bridge couldn't be found, even not after entering the IP by hand. I tried to get to the online interface so the 'http://{IP-Adress}/hue/linkbutton' but the page can't be loaded

Mevel, 2020-08-30:
start with sudo and - - debug and show output here, emulator does not startup

kompas1, 2020-08-30:
@Mevel How do you mean? sudo systemctl status hue-emulator.service was the comment I used to get this output

Mevel, 2020-08-31:
Navigate to the emulator folder and start the emulator script manually with sudo and - - debug argument

kompas1, 2020-09-01:
so this is the output if i manually start the script with sudo:

2020-09-01 11:07:10,755 - root - INFO - Using Host 192.168.178.40:80
2020-09-01 11:07:10,785 - root - INFO - b827eb9502ab
2020-09-01 11:07:10,785 - root - INFO - IP range for light discovery: 0-255
2020-09-01 11:07:10,786 - root - INFO - 127.0.0.1
2020-09-01 11:07:10,786 - root - INFO - Online Discovery/Remote API Enabled!
2020-09-01 11:07:10,792 - root - INFO - Config loaded
2020-09-01 11:07:10,800 - root - DEBUG - starting ssdp...
2020-09-01 11:07:10,801 - root - DEBUG - start ssdp broadcast
2020-09-01 11:07:10,804 - root - INFO - sync with lights
2020-09-01 11:07:10,807 - root - INFO - Starting httpd...
2020-09-01 11:07:10,828 - root - INFO - Starting ssl httpd...
2020-09-01 11:07:10,830 - urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): discovery.diyhue.org:443
2020-09-01 11:07:11,047 - urllib3.connectionpool - DEBUG - https://discovery.diyhue.org:443 "POST / HTTP/1.1" 200 6
{"ok"}

kompas1, 2020-09-01:
but if I want to start it as a service it's not working as already described

ryancasler, 2020-09-03:
Did you modify the systemd service file in some way or install the software in a non-standard location?  When you try and start the systemd service, what is output in the logs?  You can see the status by the command `sudo systremctl status hue-emulator.service`

kompas1, 2020-09-03:
@ryancasler I put the output of the systemctl status up at the front of the issue. But no I didn't modified anything. I installed it in the home directory, as much as i know that's the correct location, correct me if I'm wrong with that

ryancasler, 2020-09-04:
> I installed it in the home directory, as much as i know that's the correct location, correct me if I'm wrong with that

Yup...that is the wrong directory.  You didn't install it via the install script did you?  If you had, it would be located in /opt/hue-emulator.  That is the directory that the SystemD service file is going to try and run.  So, if the file is not located there, then the service is never going to work.  Try installing via the install script and it should work.

kompas1, 2020-09-04:
well yeah I did install via the install script. But in /opt/hue-emulator there is the HueEmulator3.py and the rest, so the cert.pem config.json and so on

kompas1, 2020-09-04:
Okay I got it. I just run the script in the /opt directory and everything is working fine. Will close the issue now

closed 2020-09-04 by kompas1