diyhue/Lights #98: nrf24 support

Shaggy013, 2020-11-06:
Hey all ,added nrf24 support 

is it possible to add lightname for each bulb or switch ?

https://github.com/Shaggy013/Lights/tree/RF24

mariusmotea, 2020-11-14:
Hi,

Can you describe what ligths are you able to control with nrf24 module?

Shaggy013, 2020-11-14:
thnx for reply 

i now only adjusted the Generic_ON_OFF_device_433Mhz ino file so it works with a nrf24l01 network,  i have now now 10 nodes and 19 switches running , but you can add the other hue options as well it is just how you would it send it over . only what the esp does is translate the lightnumber to a node nr and pin nr wich need another state and it gets a call back when node received the message 

the link to a rf24 example is in post above and works "well at my home at least :)"

there is even a python lib for nrf24
https://github.com/nRF24/RF24/tree/master/pyRF24 

so you can hook a nrf24l01 module to a spi bus and use that pyhton lib to communicate , i was just thinkering how i could adapt the mqtt.py to use the rfl24 libs instead of mqtt , seems to me only translating what diyhue sends to or wants from the scripts as on/off ,ct ,bri etc to a payload of max 144bytes .

mariusmotea, 2020-11-15:
If you want to communicate with nrf24 directly from Raspberry PI SPI Bus (hope i understood correctly) you need to create a new `protocol`, with is a pythton file that receive the light values using hue api protocol. In this file you need to have 3 functions: set_light (mandatory), get_light_state (obtional) and discover (obtional). The first one is mandatory because is the one that receive the light data and translate it to original light api. Here you can also use nrf24 library. get_light_state is only used to fetch the lights state, because some times the light is changing from 3rd party switches/ applications. Discover function is used to automatically detect and import the lights.

Shaggy013, 2020-11-15:
Yeah indeed just a new protocol , the set and get is not the hardest would something in the mind of  
set_light(address, light, data):
payload = light+data 
network.write(header(address),&payload,sizeof(payload))
...........
 and a
get_light_state(address, light):
network.read(header,&payload,sizeof(payload));
header=adress // node id 
....... etc

just getting how the discovery works , but i get the idea how diyhue works 
i am not a programmer so it take a while to see the light in the code :)

mariusmotea, 2020-11-16:
the nrf24 support may allow control for milight bulbs directly from raspberry pi host and this may work well with fast updates (entertainment mode).

Shaggy013, 2020-11-16:
that was the idea only the milight rpi lib is little outdated and thought a nrf24 network was a good stepup to it .
and the original milight diy was i thought on a rpi 
and a expansion board like this and you get them all wifi and bluetooth from the soc  and zigbee,milight,nrf24 from the spi bus :)
https://www.tindie.com/products/boros/borosrf2-dual-nrf24l01-phathat-rtc-for-pis/

mariusmotea, 2020-11-17:
Thje issue with nrf24  is the lack of protocol. Every vendor has his own binary api so for every vendor it will require a new python file in protocols folder. I think there is no issue if milight library is outdated if is working with python3. The milight protocol is the same for very long time.

Shaggy013, 2020-11-17:
like this ? 
https://github.com/dmkent/pymilight

Shaggy013, 2020-11-21:
And that is the profit of the nrf24l01 protocol you can even trick a pl1167 ;)
it is like if you have milight and also a nrf24l01  

and i sure no python programmer but the above runs in python3 with little changes in the cython files  and can already switch my lights on and off and instead of the mqtt glue it needs a diyhue glue to talk to the milightcontroller.py 

and sure it can even be more pythonized but it is a begin.

and a few more if rules to have both milight and nrf24l01 in same protocol :)

mariusmotea, 2020-11-21:
is not hard to add a new light protocol, just look at milight, esphome implementation. Hard part is to implement the 2.4GHZ api of the light

Shaggy013, 2020-11-21:
realy  ?

closed 2020-12-04 by stale[bot]