diyhue/Lights #101: ESP32

tomfrenzel, 2020-11-23:
Can you make an ESP32 compatible script for generic Lights?

ryancasler, 2020-11-24:
Generic Dimmable light or on/off light?  The on/off light would be easy to adapt to the ESP32.  The dimmable light is going to be a little harder.  The ESP32 doesn't support analogWrite, you have to use ledc, which would mean the entire sketch would have to be re-written.  It can be done but it would be much easier to just use an ESP8266 based board instead.  They are plenty powerful enough for this application.  The ESP32 is kinda overkill.

tomfrenzel, 2020-11-24:
Alright i'll use the ESP8266 then. And just to make sure:
I need to upload the Generic On/OFF Scetch **and** flash upload some kind of firmware? Because i couldn't find one for Generic lights in your repo.

ryancasler, 2020-11-24:
Are you trying to use it for an ON/OFF light or a dimmable light?  It depends on what you are trying to build.

tomfrenzel, 2020-11-24:
Just as an ON/OFF Light

ryancasler, 2020-11-24:
Then the On/Off sketch is what you want.

tomfrenzel, 2020-11-24:
And what bin File do i have to flash?

tomfrenzel, 2020-11-25:
@ryancasler Alright i uploaded the sketch and configured the esp8266 over its web interface. But my Hue Bridge (Gen. 2) can't find it. What did i do wrong?

ryancasler, 2020-11-29:
You can't use these sketches with a genuine Hue Bridge.  Only with DIYHue.

alexyao2015, 2020-12-01:
Esphome. Plan to depreciate the old sketches.

closed 2020-12-01 by alexyao2015