diyhue/Lights #102: Wifi Issue ESP8266

Badn1nja, 2020-11-28:
Hi,
I've had this as an ongoing issue I've not seen it show up before on github.

I have a esp light strip but the wifi network it creates never disappeared once its connected to my wireless network its showing as ESP-70e971 and can be connected to by anyone.

Is there a way to make this go away?

mariusmotea, 2020-11-29:
Hi,

This is an issue with new esp library. Need to declare the wifi mode in setup phase. I will create a fix for all lights soon.

mariusmotea, 2020-11-29:
Can you please test now and confirm? I made commits for all lights.

Badn1nja, 2020-11-29:
I will test shortly

mariusmotea, 2020-12-01:
It is workinking as expected now?

Badn1nja, 2020-12-02:
Yes perfect thanks for that, just had to setup Arduino on a new computer before testing

closed 2020-12-03 by mariusmotea