diyhue/Lights #105: Any chance to use ESP-01S RGB led ring?

jarno83, 2020-12-24:
Hi, I have multiple of them 16 led and 8 led ones. https://www.aliexpress.com/item/4001135425711.html
I got firmware flashed to esp01s and app found the device but didn't control the leds. I used Generic_WS2812_Strip.ino.
I think it's something to do with NeoPixelBus default pin on esp8226?

Thank you :)

Mevel, 2020-12-24:
Default Pin is the RX pin.
According to the Picture it should be Pin 7 (GPIO3)
You can edit the pin in the web Ui of your light.

https://diyprojects.io/media/2020/09/esp01-esp01s-pinout-esp8266ex.jpg

jarno83, 2020-12-25:
Hi, I checked with tester and ESP-01S-RGB  uses GPIO2. In the preferences I don't see where I can change PIN for the neopixels. 
[image]
I can see where I can turn on HW buttons, but thats it.
Can I change the output pin? Or I need to do some hw modding :)

Thanks.

Mevel, 2020-12-28:
sry my bad!
you need to specify the Pin upon sketch compile.

closed 2021-06-26 by stale[bot]